intel/zephyr.js #506: [PnP]ROM and RAM of ARC increasing obviously(above 10%) on Arduino 101 & FRDM_K64F in master-a162db1

hanyia, 2016-12-06:


hanyia, 2016-12-06:
ROM&RAM of ARC increasing in branch master:a162db1 , detail data as below table: 
<table class="tableizer-table">
<thead><tr class="tableizer-firstrow"><th>Category</th><th>Device</th><th>KPIs</th><th>WW 50master:a162db1</th><th>WW49 master:5844f9f</th><th>WW50 vs WW49</th></tr></thead><tbody>
 <tr><td>Footprint</td><td>Arduino101</td><td>Zephyr_Helloworld_ARC_ROM(Byte)</td><td>16016</td><td>11352</td><td>41.09%</td></tr>
 <tr><td>&nbsp;</td><td>&nbsp;</td><td>Zephyr_Helloworld_ARC_RAM(Byte)</td><td>5496</td><td>4900</td><td>12.16%</td></tr>
 <tr><td>&nbsp;</td><td>&nbsp;</td><td>Zephyr_I2C_ARC_ROM(Byte)</td><td>18972</td><td>14512</td><td>30.73%</td></tr>
 <tr><td>&nbsp;</td><td>&nbsp;</td><td>Zephyr_I2C_ARC_RAM(Byte)</td><td>5584</td><td>4980</td><td>12.13%</td></tr>
 <tr><td>&nbsp;</td><td>&nbsp;</td><td>Zephyr_WebBluetoothGroveLcdDemo_ARC_ROM(Byte)</td><td>22412</td><td>19388</td><td>15.60%</td></tr>
 <tr><td>&nbsp;</td><td>&nbsp;</td><td>Zephyr_WebBluetoothGroveLcdDemo_ARC_RAM(Byte)</td><td>6024</td><td>5412</td><td>11.31%</td></tr>
 <tr><td>&nbsp;</td><td>FRDM_K64F</td><td>Zephyr_Helloworld_ARC_ROM(Byte)</td><td>16016</td><td>11352</td><td>41.09%</td></tr>
 <tr><td>&nbsp;</td><td>&nbsp;</td><td>Zephyr_Helloworld_ARC_RAM(Byte)</td><td>5496</td><td>4900</td><td>12.16%</td></tr>
</tbody></table>

jimmy-huang, 2016-12-06:
> Category 	Device 	KPIs 	WW 50master:a162db1 	WW49 master:5844f9f 	WW50
> Footprint 	Arduino101 	Zephyr_Helloworld_ARC_ROM(Byte) 	16016 	11352
> 		Zephyr_Helloworld_ARC_RAM(Byte) 	5496 	4900 	12.16%
> 		Zephyr_I2C_ARC_ROM(Byte) 	18972 	14512 	30.73%
> 		Zephyr_I2C_ARC_RAM(Byte) 	5584 	4980 	12.13%
> 		Zephyr_WebBluetoothGroveLcdDemo_ARC_ROM(Byte) 	22412 	19388 	15.60%
> 		Zephyr_WebBluetoothGroveLcdDemo_ARC_RAM(Byte) 	6024 	5412 	11.31%
> 	FRDM_K64F 	Zephyr_Helloworld_ARC_ROM(Byte) 	16016 	11352 	41.09%

The increase of ROM/RAM is due to the updated in Zephyr's unified 
kernel, since the ARC used to build with nano kernel, and now since nano 
and micro kernels are now merged in Zephyr, and there's no option to 
build the smaller nano kernel, we are basically now building ARC with 
microkernel-equivalent instead, thus the increase in ROM and RAM size.

Jimmy

grgustaf, 2016-12-07:
@hanyia The ARC build is not applicable to the FRDM_K64F, it's specific to the Arduino 101. So that's why those lines in your table come out identical. They are meaningless.

@jimmy-huang's explanation makes sense, so we're unfortunately just going to have to live with this increase. This is the new baseline.

Thanks for the good work though, it's nice to catch this kind of change and quantify it.

hanyia, 2016-12-07:
@grgustaf @jimmy-huang  thanks your explanation, I will remove ARC to the FRDM_K64 and set this as new baseline.

closed 2016-12-07 by hanyia