intel/zephyr.js #513: Installation guidance, post SDK setup: "OS X is not supported!"

rwaldron, 2016-12-06:
I've looked through the existing pull requests and issues and based on what I reviewed, I'm still stuck at the following: 

1. If I follow the instructions in the readme, I will not get past this step: https://github.com/01org/zephyr.js#install-the-zephyr-sdk 
1. If installing the SDK fails on OSX, then how does one get to this step: https://github.com/01org/zephyr.js/pull/189 ? That is, running `make linux`?
    1. Am I misunderstanding the role of the SDK? 

```
rwaldron in ~/Downloads
$ chmod +x zephyr-sdk-0.8.2-i686-setup.run
rwaldron in ~/Downloads
$ sudo ./zephyr-sdk-0.8.2-i686-setup.run
Password:
Verifying archive integrity... All good.
Uncompressing SDK for Zephyr  100%
OS X is not supported!
```

grgustaf, 2016-12-06:
Sorry, I haven't used OSX, so I'm not sure if that's new or something to be overlooked.

@poussa, can you look at this?

poussa, 2016-12-07:
Okay, we have multiple issues here but lets start fixing them one by one.

First, our (zephyr.js project) Zephyr SDK installation document (https://github.com/01org/zephyr.js#install-the-zephyr-sdk) is incomplete. It only talks about Linux environment without explicitly stating that. We should point in our documentation (the SDK installation part) to Zephyr project docs and not to duplicate that.

Next, the Zephyr SDK installation for Mac are here: https://www.zephyrproject.org/doc/getting_started/installation_mac.html# . You need to build the cross compiler (crosstool-ng) yourself and set the env variables to point to that, like:
```
ZEPHYR_SDK_INSTALL_DIR=/Volumes/CrossToolNG/x-tools
ZEPHYR_GCC_VARIANT=xtools
```

Finally, when you type make (on zephyr.js), we have an error:
```
[spoussa@zephyr.js (master *)]$ make
Using module: Console
ccflags-y += -DBUILD_MODULE_CONSOLE
sed: 1: "src/Makefile": unterminated substitute pattern
make: *** [analyze] Error 1
```

This is something we (zephyr.js project) should fix asap. I expect we have few more down the road since Mac has not been used lately and lot of changes has been merged. But as I said, lets fix the issues one by one and get the Mac support back in

poussa, 2016-12-07:
@jprestwo Can you make the linux build to optionally include the OCF dependencies. In order to make linux to work on Mac we need to do several fixes for iotivity-constrained and CBOR. But to get basic functionality available on Mac it would be good to have a way to exclude the OCF deps from the Linux (and OSX) build.

jprestwo, 2016-12-14:
https://github.com/01org/zephyr.js/pull/541 changes the Linux build to only include iotivity-constrained on Linux. I have no way of testing it on Mac so if @poussa or anyone else with a Mac could test out the PR that would be great!

jprestwo, 2016-12-22:
@poussa, is this fixed now? It should be in combination between my commit and yours.

poussa, 2017-01-04:
This is fixed now. The next step is to enable more modules on OSX like OCF.

closed 2017-01-04 by poussa

bsingh5266, 2021-07-13:
Getting started with Zephyr (Setup toolchain, compilation, board bring-up, Hello World example)!!

Systems: MacBook
Board     : nRF52833 DK (Optional)

https://youtu.be/4emoE-8hsqg