intel/zephyr.js #514: [REG] D0 and D1 cannot be used as output pins on FRDM_K64F

xiaolongx, 2016-12-07:
Flashing [test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js) on FRDM_K64F with commit a162db1, D0 and D1 cannot be used as output pins.

Output result:
```
Test K64f_pins APIs
PASS - K64f_pins: D0 defined
PASS - K64f_pins: D0 input
FAIL - K64f_pins: D0 output
PASS - K64f_pins: D1 defined
PASS - K64f_pins: D1 input
FAIL - K64f_pins: D1 output
```
Branch | Build ID | Date | Result
---|---|---|---
master | 5844f9f | Dec 7, 2016 | PASS
master | a162db1  | Dec 7, 2016 | FAIL

qiaojingx, 2017-02-15:
Another issue #665  is reproduced on FRDM-K64F with commit 3f8eb85.
Will verify this issue after #665 fixed.

grgustaf, 2017-03-24:
I can reproduce that D0 and D1 don't work as inputs, but e.g. D3 does.

I also so far can't seem to use any of the pins as an output. I haven't tried this for a while. I guess I'll look into the two commits you list.

grgustaf, 2017-03-31:
Fixed by #954.

qiaojingx, 2017-04-01:
Verified with commit f3d587d + #954. This issue is not reproducible.
Will close it after patch merged.

PR #954 (2017-04-03): [k64f] Restore GPIO functionality by enabling pins via pinmux
Zephyr defaults changed around January for this board to no longer
set GPIO as default pinmux state for any of the external pins. So we
need to override them ourselves. Eventually we may need to get fancier
to set pinmux based on what the user's script wants to use the pins
for, or set up a default "ABI" for pinmux settings for the board. The
pinmux must be set up at board initialization time apparently.

Add new zjs_pinmux.c (currently for k64f only) to configure the
pinmux settings at initialization time.

Add a new prj.conf to en [... cut]
commit 9e12add8 (2017-04-03): [k64f] Restore GPIO functionality by enabling pins via pinmux (#954)

Zephyr defaults changed around January for this board to no longer
set GPIO as default pinmux state for any of the external pins. So we
need to override them ourselves. Eventually we may need to get fancier
to set pinmux based on what the user's script wants to use the pins
for, or set up a default "ABI" for pinmux settings for the board. The
pinmux must be set up at board initialization time apparently.

Add new zjs_pinmux.c (currently for k64f only) to configure the
pinmux settings at initialization time.

Add a new prj.co [... cut]
closed 2017-04-03 by jimmy-huang