intel/zephyr.js #524: [REG] Buffer tests no longer work

brianjjones, 2016-12-09:
The files tests/test-buffer.js and tests/test-buffer-rw.js no longer run, I just get a "JerryScript: cannot run javascript" error.  This was tested on master using both an Arduino101 and FRDM-K64 board.

xiaolongx, 2016-12-09:
This issue also occurs on samples/Buffer.js and samaples/tests/BufferMemory.js using both Arduino101 and FRDM_K64F.

Branch | Build ID | Result
---|---|---
master | d4e1d64 | PASS
master | dc703a1 | FAIL

grgustaf, 2016-12-09:
@xiaolongx, these tables of last known PASS and FAIL are very helpful, thank you. If you guys are ever bored, you're welcome to git bisect to find the exact patch that broke the condition. :) But again this is awesome and lets me easily do that.

grgustaf, 2016-12-13:
Hmm, interesting that you filed this, @brianjjones, because I just git bisected this to your commit dc703a1. :)

grgustaf, 2016-12-13:
Fixed with #535.

xiaolongx, 2016-12-14:
This issue cannot reproduce on both Arduino 101 and FRDM-K64F after #535 merged.

closed 2016-12-14 by grgustaf