intel/zephyr.js #531: [Sensor] The event.reading.illuminance is too low in AmbientLight.js

xiaolongx, 2016-12-12:
Flashing [AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js) with commit 31c3908 on Arduino 101, the event.reading.illuminance is too low even using flashlight.
```
Ambient light test...
zjs_sensor_create: unsupported frequency, defaulting to 50Hz
state: activating
ipm_console0: 'start ambient light 12'
state: activated
(very dark): 0.00019422420860140712
(very dark): 0.0001910913093819011
(very dark): 0.00019422420860140712
```

jimmy-huang, 2017-03-14:
@xiaolongx, @qiaojinxgis this still reproducible?

qiaojingx, 2017-03-15:
Verified with commit 9777997 on Arduino 101. 
This issue will not be reproduced if wire sensor's power to Arduino 5V.
[image]

But if wire sensor's power to Arduino 3.3V. The event.reading.illuminance is still too low.
[image]

jimmy-huang, 2017-03-16:
Ok, this make sense, I'll probably update the wiring instructions in the same to attach to sensor only to 5v instead.