intel/zephyr.js #544: [aio] ADC driver in Zehpyr may not work with frdm-k64f board

jimmy-huang, 2016-12-14:
I started porting AIO to suport non-ipm boards like frdm-k64f, but I am running into an issue that the default "ADC_0" driver doesn't work on the frdm-k64f, so when I call device_get_binding("ADC_0"), I get a NULL value.  Someone from the team will probably have to look into this with the Zephyr team to see if ADC is actually supported, and can pick up from where I left.

https://github.com/jimmy-huang/zephyr.js/tree/k64f-aio

grgustaf, 2017-04-12:
We learned that it's not supported currently but is being tracked here in [Zephyr JIRA 1396](https://jira.zephyrproject.org/browse/ZEP-1396).

grgustaf, 2017-08-23:
This has apparently been merged so we should make it work for 0.5.

qiaojingx, 2018-02-06:
Basic feature is ready on FRDM-K64F. Verified with samples([AIO.js](https://github.com/01org/zephyr.js/blob/master/samples/AIO.js)) and tests([test-aio.js](https://github.com/01org/zephyr.js/blob/master/tests/test-aio.js)).
Commit ID: 4cf09c5 in zjs-0.5 branch
Platform: FRDM-K64F.
ZephyrSDK: 0.9.2
Zephyr: v1.10 release