intel/zephyr.js #548: [mod] Error thrown when import custom module

cuiyanx, 2016-12-15:
When import custom module in another custom module, throw out error. 
e.g.  import [modules/GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/modules/GroveLCD.js) module in the [modules/BIM280.js](https://github.com/01org/zephyr.js/blob/master/modules/BMP280.js) module:

```
    // API object
    var bmp280API = {};

    var GroveLCD= require("GroveLCD.js");
    var i2c = require("i2c");
    bmp280API.i2cDevice = i2c.open({ bus: 0, speed: 100 });
```
error log:
```
native_require_handler: module not found
JerryScript : cannot run javascript
```

grgustaf, 2016-12-15:
@brianjjones I bet this is my fault with a recent patch or true where I tried to simplify some modules stuff. Could you take a look, though, since I'm not too familiar with your multiple .js support?

Also, we should add a test case using multi-js to Travis to prevent this next time, can you maybe give me a suggested representative build command or two?

brianjjones, 2016-12-16:
@grgustaf @cuiyanx Actually this simply isn't supported at this time.  Modules can't rely on other current modules currently.  The only javascript that can 'require' a javascript file is the main javascript declared by JS=*.  I don't think this will be hard to add, but I need to add it.  I'll make sure and tackle this tomorrow.

@grgustaf  To be clear this doesn't mean modules is broken, he simply found a use case I didn't support.  For example, samples/I2CMBMP280.js still works.

grgustaf, 2016-12-21:
Thanks, I've changed it to a feature request and updated the title.

brianjjones, 2017-04-05:
This will be supported by ashell when https://github.com/01org/zephyr.js/pull/959 is accepted.

However, it will still not be supported by the regular ZJS build.  I'll need to integrate these changes into it.

cuiyanx, 2017-09-01:
Fixed and closed

closed 2017-09-01 by cuiyanx