intel/zephyr.js #549: [sensor] Error throw when import costom module or large segment code

cuiyanx, 2016-12-15:
When add a large segment code or impoet costom module [BMP280.js](https://github.com/01org/zephyr.js/blob/master/modules/BMP280.js) in samples/Accelerometer.js, throw out error.

error log:
```
zjs_sensor_ipm_send_sync: FATAL ERROR, ipm timed out
zjs_sensor_create failed to init
```

grgustaf, 2016-12-15:
@brianjjones, might be wise to git bisect this one and see where it broke. Might have been one of @jimmy-huang's recent patches or my fault again.

brianjjones, 2016-12-16:
@grgustaf This is a race condition.  By adding the BMP280.js, we now have two things trying to init at the same time.  Apparently IPM can't handle this currently, and the sensor IPM times out.  If you wrap the entire Accelerometer.js code with a setTimeout for a second, it works fine.  This may just be a limitation of the IPM system, but perhaps we can do a better job with scheduling them.  I'll dig in more and let you know.

brianjjones, 2016-12-16:
This is solved by https://github.com/01org/zephyr.js/pull/555

xiaolongx, 2016-12-20:
This issue cannot reproduce on Arduino 101 with commit 5a25755

grgustaf, 2016-12-21:
Closing, thanks.

closed 2016-12-21 by grgustaf