intel/zephyr.js #551: [I2C] Fail to throw error when opening an invalid bus on FRDM-K64F

xiaolongx, 2016-12-15:
Flashing [test-i2c.js](https://github.com/01org/zephyr.js/blob/master/tests/test-i2c.js) on FRDM-K64F, fail to throw error when opening an invalid bus.
```
expectThrow("I2C: open invalid bus", function () {
    i2c.open({ bus: 1, speed: 100 });
});
```

Error logs:
```
I2C bus I2C_1 is not a valid I2C bus.
FAIL - I2C: open invalid bus
```

Branch | Build ID | Target Device | Date | Result
---|---|---|---|---
master | 5c969a5 | Arduino 101 | Dec 15, 2016 | PASS
master | 5c969a5 | FRDM-K64F | Dec 15, 2016 | FAIL

grgustaf, 2016-12-15:
I'm not sure if this a regression or it never threw an error? This may come down to the expectation that JS will throw for unlikely errors. It may well be fixed just by moving to the new APIs, since they probably specify this behavior more clearly.

brianjjones, 2016-12-15:
On the Arduino 101 bus 1 is valid (at least in theory I haven't tried in practice)

I'll check this out today and if it is indeed invalid I'll change it from only printing an error to throwing

brianjjones, 2016-12-15:
This issue will be fixed by https://github.com/01org/zephyr.js/pull/554

xiaolongx, 2017-01-06:
This issue cannot reproduce on FRDM-K64F with commit 2345f96, close

closed 2017-01-06 by xiaolongx