intel/zephyr.js #558: [infrastructure] Cache SDK in Travis CI

baleboy, 2016-12-19:
Installing the Zephyr SDK could be cached to speed up CI run times.

baleboy, 2016-12-19:
@jprestwo I think you were playing with this at some point, are you still working on it? If not I'd like to give it a shot.

grgustaf, 2016-12-19:
Based on this line from the [Travis Caching](https://docs.travis-ci.com/user/caching/#Things-not-to-cache) page under "Things Not to Cache", I don't believe this is going to be too much help.

Large files that are quick to install but slow to download do not benefit from caching, as they take as long to download from the cache as from the original source.

But I guess it's not that quick to install, maybe it would help. It also occurs to me maybe we could pass flags to the SDK installer to leave out cross-compilers that we're not going to be using? That could speed the install up quite a bit since it installs like six and we probably use two?

baleboy, 2016-12-20:
When I first set up Travis I had the cache somewhat working, and it was making a difference (seconds vs minutes), perhaps because downloading from Travis' own infrastructure is faster than Zephyr's. However the cached file disappeared at some point and I couldn't figure out why, so I decided not to use the cache.

Anyway I'll try and report my findings.

grgustaf, 2017-04-12:
I finally did this in PR #67dd.

closed 2017-04-13 by grgustaf