intel/zephyr.js #560: Some event functions return a boolean instead of 'this'

jprestwo, 2016-12-19:
@grgustaf pointed out that there are some event API's that never made the full conversion to returning 'this' and still return a boolean. The event API's should return 'this' as is the Node spec.

grgustaf, 2016-12-20:
remove_listener and at least one other function return bool or this. get_listener_count returns a number or this. So we need to look across all the functions here I think, I've just stumbled across these. But seems like it would be a problem.

grgustaf, 2016-12-20:
#559 is kind of related, if we documented what they were supposed to return it would be more obvious when they don't. :)

jprestwo, 2016-12-22:
This should now be fixed with https://github.com/01org/zephyr.js/pull/561

closed 2017-01-04 by grgustaf