intel/zephyr.js #562: Error thrown after finish test-grovelcd-manual tests on FRDM-K64F

xiaolongx, 2016-12-20:
Error thrown after running [test-grovelcd-cursor-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-cursor-manual.js), [test-grovelcd-ds-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-ds-manual.js), [test-grovelcd-rgb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-rgb-manual.js) on FRDM-K64F, but works well on Arduino 101.

Error logs:
```
test RGB color
setColor(R, G, B): expected result 255, 0, 0
color gradient:  red
setColor(R, G, B): expected result 0, 255, 0
color gradient: green
setColor(R, G, B): expected result 0, 0, 255
color gradient:  blue
setColor(R, G, B): expected result 255, 255, 0
color gradient: yellow
setColor(R, G, B): expected result 255, 0, 255
color gradient: purple
setColor(R, G, B): expected result 0, 255, 255
color gradient:  cyan
setColor(R, G, B): expected result 255, 255, 255
color gradient: white
Testing completed
***** Reserved Exception ( -16) *****
Fatal fault in essential thread! Spinning...
```

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 5a25755 | Dec 20, 2016 | PASS
master | FRDM-K64F | 5a25755 | Dec 20, 2016 | FAIL

brianjjones, 2016-12-21:
I confirm this is an issue, and it happens when clearInterval is called.  The clearInterval function completes, but it crashes shortly after.  What is strange is that code isn't different in any way between the two boards. If I turn the test into a while loop rather than a setInterval it has no issue.  I'll keep investigating into what the root cause is.

brianjjones, 2016-12-30:
I found the problem in zjs_callbacks, basically after the callback is called there is a chance that it was deleted, which is the case with this test.  It doesn't appear to hurt the Arduino, probably due to some difference of the order of when the zjs_remove_callback deletes cb_map[id].

I made a fix https://github.com/01org/zephyr.js/pull/582

xiaolongx, 2017-01-09:
This issue cannot reproduce on FRDM-K64F with commit 2345f96, close

closed 2017-01-09 by xiaolongx