intel/zephyr.js #566: [Console] Fail to throw exception when output an undefined value

xiaolongx, 2016-12-21:
If we only run `console.log(a);`, it will output `JerryScript: cannot run javascript` due to 'a' is undefined. But adding `console.log(a)` in setTimeout() and setInterval(), no exception thrown on both Arduino 101 and FRDM-K64F.
```
setTimeout(function() {
    console.log(a);
}, 1000);
```
But using the same code on chrome, it will throw exception like this: 
`Uncaught ReferenceError: a is not defined`

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino | 5a25755 | Dec 21, 2016 | Nothing Output
master | FRDM-K64F | 5a25755 | Dec 21, 2016 | Nothing Output

grgustaf, 2016-12-21:
Maybe this is a good thing. ;) Worth looking into, maybe we're squashing all exceptions somehow?

jprestwo, 2017-01-23:
I do see this behavior, whether or not its a bug is up in the air. I think it could be a potential JerryScript issue. The reason that calling `console.log(a)`  (not in `setTimeout()`) fails is because the JerryScript parser is catching it before execution. In the Timer module, functions get called later in the callback module. I looked in callbacks and we are checking to see if there was an error in execution of the callback function:
```
ret_val = jerry_call_function(cb_map[id]->js_func, cb_map[id]->this, data, sz);
if (jerry_value_has_error_flag(ret_val)) {
    DBG_PRINT("callback %d returned an error for function\n", id);
}
```

The question is, are we doing this correctly? I am not sure about that. This is how we do it everywhere else in the code, but perhaps this check won't catch syntax/parse/reference errors. The return value may just be related to errors thrown during runtime.

jprestwo, 2017-03-02:
@grgustaf, can this be closed now that we added JS error printing to callbacks/main. Its the best we can do given the situation, since there is no way to actually throw an error (since its in the main loop).

qiaojingx, 2017-03-09:
Verified with commit b29dbbc on Arduino 101. 
Run: `console.log(a);`
Output: 
```
src/main.c:228 main():                                                          
[ERROR] Error running javascript                                                
                                                                                
src/zjs_util.c:381 zjs_print_error_message():                                   
[ERROR] Uncaught exception: Error: 
```
Run: 
```
setTimeout(function() {
    console.log(a);
}, 1000);
```
Output:
```
src/zjs_util.c:381 zjs_print_error_message():                                   
[ERROR] Uncaught exception: Error: 
```

xiaolongx, 2017-03-14:
Closed since JS error printing is added to callbacks/main

closed 2017-03-14 by xiaolongx