intel/zephyr.js #579: [REG] Fail to find modules via native_require_handler

xiaolongx, 2016-12-27:
Flashing [I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js), [test-uart.js](https://github.com/01org/zephyr.js/blob/master/tests/test-uart.js) on Arduino 101, the samples cannot be executed.  This issue occurs after commit bbd6804 merged.
Source Code
```
console.log("Starting I2C demo using BMP280 and Grove LCD");

/* Globals */

// Import JavaScript modules from 'modules' folder
var bmp280 = require("BMP280.js");
var glcd = require("GroveLCD.js");
```
Error logs
```
ipm_console0: 'Sensor core running ZJS ARC support image'

Starting I2C demo using BMP280 and Grove LCD
native_require_handler: module not found

src/main.c:200 main():
[ERROR] JerryScript: cannot run javascript
```
Branch | Build ID | Date | Result
---|---|---|---
master | e0d4559 | Dec 27, 2016 | FAIL
master | bbd6804 | Dec 27, 2016 | FAIL
master | 5a25755 | Dec 27, 2016 | PASS

grgustaf, 2017-01-05:
Also fixed by #584.

xiaolongx, 2017-01-09:
This issue cannot reproduce on Arduino 101 with commit 2345f96, close

closed 2017-01-09 by xiaolongx