intel/zephyr.js #580: Fail to output long string on FRDM-K64F

xiaolongx, 2016-12-27:
Flashing [Console.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/Console.js) on FRDM-K64F with commit e0d4559, the long string can not be displayed properly, but works well on Arduino 101.
Source Code:
```
console.log("\nTesting a long string...");
var longstr =
    "123456789012345678901234567890123456789012345678901234567890" +
    "123456789012345678901234567890123456789012345678901234567890" +
    "123456789012345678901234567890123456789012345678901234567890" +
    "123456789012345678901234567890123456789012345678901234567890" +
    "123456789012345";  // length 255
console.log(longstr);
```

Output Result:
```
Testing a long string...
123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345675
```

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino 101 | e0d4559 | Dec 27, 2016 | PASS
master | FRDM-K64F | e0d4559 | Dec 27, 2016 | FAIL

grgustaf, 2017-01-04:
Did this work on older builds on K64F?

xiaolongx, 2017-01-19:
It also did not work on older commit, I have updated the issue, it should be related with minicom command line.

xiaolongx, 2017-02-14:
The same issue is reproduce on Arduino 101 with commit 3f8eb85. Update the title.

grgustaf, 2017-04-07:
I forgot to comment here, but I tested this and it worked for me a few days ago on k64f.

grgustaf, 2017-04-07:
Verified again just now with master. Please retest.

grgustaf, 2017-04-07:
Note: It's 255 characters, so it results in 3 80-char lines of output plus fifteen characters on the fourth line: 3*80+15 == 240 + 15 == 255.

Just in case was confusing someone because there are 4+1 lines in the source (4*60+15 == 255)

qiaojingx, 2017-04-10:
Verify with commit cc2311a on arduino 101 and FRDM-K64F. This issue is not reproducible. Closed.

closed 2017-04-10 by xiaolongx