intel/zephyr.js #587: link error when build JS for 96b_carbon

tidyjiang8, 2017-01-06:
Hello, 

There is a 96b_carbon in my hand, and I want to try running zephyr.js on it.

But when I build, it prompts the following:
```
  AR      libzephyr.a
  LINK    zephyr.lnk
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/libexec/arm-poky-eabi/gcc/arm-poky-eabi/5.2.0/real-ld: libjerry-core.a(jerry-all-in.c.obj): Relocations in generic ELF (EM: 6)
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/libexec/arm-poky-eabi/gcc/arm-poky-eabi/5.2.0/real-ld: libjerry-core.a(jerry-all-in.c.obj): Relocations in generic ELF (EM: 6)
...
/opt/zephyr-sdk/sysroots/i686-pokysdk-linux/usr/libexec/arm-poky-eabi/gcc/arm-poky-eabi/5.2.0/real-ld: libjerry-core.a(jerry-all-in.c.obj): Relocations in generic ELF (EM: 6)
libjerry-core.a(jerry-all-in.c.obj): error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
/home/work/me/zephyr.js/deps/zephyr/Makefile:852: recipe for target 'zephyr_prebuilt.elf' failed
make[3]: *** [zephyr_prebuilt.elf] Error 1
make[3]: Leaving directory '/home/work/me/zephyr.js/outdir/96b_carbon'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/work/me/zephyr.js/deps/zephyr'
/home/work/me/zephyr.js/deps/zephyr/Makefile.inc:82: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/work/me/zephyr.js'
Makefile:53: recipe for target 'zephyr' failed
make: *** [zephyr] Error 2
```
That is, the build process is passed, but the link process, that links libzephyr.a and libjerry-core.a together, is failed. 

Could anyone give me a hand ?

**My Detail Step:**
```bash
$ cd deps/zephyr
$ git checkout master
$ git pull original master
$ cd ../..
$ git add deps/zephyr
$ git commit xxxx
$ make BOARD=96b_carbon JS=samples/HelloWorld.js
```

_Thanks._ 
Tidy

xiaolongx, 2017-01-06:
Hi Tidy

I cannot reproduce this issue with commit e034d5c and 2345f96.

Reproduce steps:
```
$ git clone https://github.com/01org/zephyr.js
$ source zjs-env.sh 256
$ make update
$ source deps/zephyr/zephyr-env.sh
$ make BOARD=96b_carbon JS=samples/HelloWorld.js
```
Result:
```
make[3]: Leaving directory '/home/ZJS/ww01/zephyr.js/outdir/96b_carbon'
make[2]: Leaving directory '/home/ZJS/ww01/zephyr.js/deps/zephyr'
make[1]: Leaving directory '/home/ZJS/ww01/zephyr.js'
```

tidyjiang8, 2017-01-06:
Hi @xiaolongx ,

Thanks for your help, I clone zephyr.js again, and it works well now.

Sorry for the trouble. 

Tidy.

closed 2017-01-06 by tidyjiang8