intel/zephyr.js #588: Fail to build with "SNAPSHOT=on"

xiaolongx, 2017-01-06:
The Snapshot build failed with command: `$ make JS=samples/AIO.js SNAPSHOT=on all` on commit 2345f96.

Error Logs:
```
==============================
Build succeeded!
==============================
gcc -Isrc/ -I/home/xiaolong/00_xiaolong/ZJS/ww01/zephyr.js/deps/jerryscript/jerry-core -L /home/xiaolong/00_xiaolong/ZJS/ww01/zephyr.js/deps/jerryscript/build/lib/ -static -o /home/xiaolong/00_xiaolong/ZJS/ww01/zephyr.js/outdir/snapshot//snapshot /home/xiaolong/00_xiaolong/ZJS/ww01/zephyr.js/outdir/snapshot//src/snapshot.o /home/xiaolong/00_xiaolong/ZJS/ww01/zephyr.js/outdir/snapshot//src/zjs_script.o -fno-asynchronous-unwind-tables -fno-omit-frame-pointer -Wno-format-zero-length -Wno-main -ffreestanding -Os -fno-stack-protector -ffunction-sections -fdata-sections -Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Wno-pointer-sign -fno-strict-overflow -Werror=implicit-int -Wall -std=gnu99 -flto -Wpointer-sign  -DZJS_LINUX_BUILD -DZJS_PRINT_FLOATS -l jerry-core -lm -pthread
/tmp/ccb5pHph.ltrans0.ltrans.o: In function `main':
<artificial>:(.text.main+0x25): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0x77): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0xc9): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0x10f): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0x14f): undefined reference to `zjs_shorten_filepath'
/tmp/ccb5pHph.ltrans0.ltrans.o:<artificial>:(.text.main+0x1a6): more undefined references to `zjs_shorten_filepath' follow
collect2: error: ld returned 1 exit status
Makefile.snapshot:84: recipe for target 'snapshot' failed
make[1]: *** [snapshot] Error 1
make[1]: Leaving directory '/home/xiaolong/00_xiaolong/ZJS/ww01/zephyr.js'
Makefile:178: recipe for target 'generate' failed
make: *** [generate] Error 2
```

grgustaf, 2017-01-06:
@brianjjones with Jimmy out could you take a look at this?

brianjjones, 2017-01-09:
Fix created https://github.com/01org/zephyr.js/pull/589

xiaolongx, 2017-01-12:
This issue cannot reproduce on Arduino 101 with commit 11a32ac(master branch) and cd4e16a(zjs-0.2 branch), close.

closed 2017-01-12 by xiaolongx