intel/zephyr.js #593: [REG] Error message "BMP280 not found! ChipId 70 != 58" displays in Serial log

xiaolongx, 2017-01-10:
The [I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js) works well on Arduino 101 with commit 2345f96, but `BMP280 not found! ChipId 70 != 58` displays in Serial log.
Error logs:
```
Starting I2C demo using BMP280 and Grove LCD
BMP280 not found! ChipId 70 != 58
```
BTW, it should throw some exception when get the wrong ChipId.
Source Code:
```
function bmp280Init() {
    var reader = bmp280.i2cDevice.read(bmp280.bmp280Addrs.ADDRESS, 1,
                                       bmp280.bmp280Addrs.REGISTER_CHIPID);

    // Check that we have a BMP280 on the I2C bus
    if (reader.toString("hex") !== "58")
        print("BMP280 not found! ChipId " + reader.toString("hex") + " != 58");
```

grgustaf, 2017-04-13:
Somehow we missed this one; no idea if it's still an issue?

qiaojingx, 2017-04-13:
Still can see this issue:
[image]

brianjjones, 2017-04-14:
It doesn't hurt anything, I probably read the spec wrong on the BMP280 chip.  Despite the error message it works fine. I'll look into whether its a valid check, and if not just delete the message.

brianjjones, 2017-08-30:
Finally getting around to fixing this with #1506

qiaojingx, 2017-09-01:
Verified with commit a0b1217 + #1506 . This issue is not reproduced.

closed 2017-09-13 by xiaolongx