intel/zephyr.js #597: [REG] Performance.now() result over known delay on Arduino 101

xiaolongx, 2017-01-12:
The test result of [test-performance.js](https://github.com/01org/zephyr.js/blob/master/tests/test-performance.js) fail on Arduino_101, but works well on FRDM-K64F. This issue should be caused by commit 47b00cc.

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | FRDM_K64F | 11a32ac | Jan 12, 2017 | PASS
master | Arduino_101 | 47b00cc | Jan 12, 2017 | FAIL
master | Arduino_101 | 5108c86 | Jan 12, 2017 | PASS
zjs-0.2 | Arduino 101 |

xiaolongx, 2017-01-18:
This issue cannot reproduce on Arduino 101 with commit c707669

xiaolongx, 2017-02-09:
This issue still can be reproduced on Arduino 101 with commit [eafdfe9](https://github.com/01org/zephyr.js/commit/eafdfe9e3f239e79e006fa7e2832eb4df5ba8305) in zjs-0.2 branch.
Reopen.

grgustaf, 2017-03-29:
@xiaolongx this seems to be passing now, can you check again?

xiaolongx, 2017-03-29:
Verified with commit 694cf15c. This issue is not reproducible.
Closed

closed 2017-03-29 by xiaolongx