intel/zephyr.js #603: [REG] Fail to run OCF samples on FRDM-K64F after zephyr and iotivity-constrained update

xiaoyu3x, 2017-01-16:
Flashing [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClient.js), [OcfClientUpdate.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfClientUpdate.js), [OcfDeviceInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfDeviceInfo.js), [OcfPlatformInfo.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfPlatformInfo.js), [OcfServer.js](https://github.com/01org/zephyr.js/blob/master/samples/OcfServer.js) on FRDM_K64F with commit c707669, OCF cannot be started after iotivity-constrained and zephyr updated. This commit also affect [test-k64f-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-k64f-pins.js), D14 and D15 cannot be used as output pins.

Serial Logs:
`assertion "handle->txBdCurrent" failed: file "/home/neusoft/zephyr.js/deps/zephyr/ext/hal/ksdk/drivers/fsl_enet.c"

Branch | Target Device | Build ID | Date | Result
---|---|---|---
master | FRDM-K64F | 2345f96 | Jan 9, 2017 | PASS
master | FRDM-K64F | e712167 | Jan 16,2017 | PASS
master | FRDM-K64F | c707669 | Jan 16,2017 | FAIL `

qiaojingx, 2017-02-15:
Ocf samples still can not run on FRDM_K64F and Linux with commit 3f8eb85.
Serial logs for each ocf sample on FRDM_K64F:
```
Started buffer handler process
```

Serial logs for OcfServer.js on Linux:
```
src/main.c:226 main():
[ERROR] JerryScript: cannot run javascript
```

xiaoyu3x, 2017-02-28:
This issue isn't reproduced and new issue tracking #759, so closed it.

closed 2017-02-28 by xiaoyu3x