intel/zephyr.js #604: [REG] commit c707669

qiaojingx, 2017-01-16:
After flashing the samples with commit c707669, four kinds of samples failed, and they were caused by commit c707669:
1. Samples:
AIO:
 [HeartRateDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js), 
I2C:
[I2C.js](https://github.com/01org/zephyr.js/blob/master/samples/I2C.js), [I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js), [GroveLCD.js](https://github.com/01org/zephyr.js/blob/master/samples/GroveLCD.js),
Sensor:
 [Gyroscope.js](https://github.com/01org/zephyr.js/blob/master/samples/Gyroscope.js),[Accelerometer.js](https://github.com/01org/zephyr.js/blob/master/samples/Accelerometer.js),[AmbientLight.js](https://github.com/01org/zephyr.js/blob/master/samples/AmbientLight.js) 
BLE:
[WebBluetoothGroveLcdDemo.js](https://github.com/01org/zephyr.js/blob/master/samples/WebBluetoothGroveLcdDemo.js) 
2. Tests:
I2C:
[test-i2c.js](https://github.com/01org/zephyr.js/blob/master/tests/test-i2c.js) 
GroveLCD:
[test-grovelcd-ds-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-ds-manual.js),[test-grovelcd-cursor-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-cursor-manual.js), [test-grovelcd-rgb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-rgb-manual.js),[test-grovelcd.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd.js), 

Serial Logs:

for other LCD cases:
`test RGB color
ipm_console0: 'Sensor core running ZJS ARC support image'
src/zjs_grove_lcd_ipm.c:41 zjs_glcd_ipm_send_sync():
[ERROR] FATAL ERROR, ipm timed out
[ERROR] zjs_glcd_call_remote_function: ipm message failed or timed ou!
src/main.c:200 main():
[ERROR] JerryScript: cannot run javascript
ipm_console0: 'Sensor core running ZJS ARC support image'`

for I2C cases:
`I2C sample...
ipm_console0: ''
ipm_console0: 'src/main.c:278 handle_i2c():'
ipm_console0: '[ERROR] i2c_write failed!'`

Failed to build sensor cases:
`/home/neusoft/00_test/zephyr.js/arc/src/main.c: In function 'convert_sensor_value':
/home/neusoft/00_test/zephyr.js/arc/src/main.c:497:10: error: 'SENSOR_VALUE_TYPE_INT' undeclared (first use in this function)
     case SENSOR_VALUE_TYPE_INT:
          ^
/home/neusoft/00_test/zephyr.js/arc/src/main.c:497:10: note: each undeclared identifier is reported only once for each function it appears in
/home/neusoft/00_test/zephyr.js/arc/src/main.c: In function 'start_accel_trigger':
/home/neusoft/00_test/zephyr.js/arc/src/main.c:693:17: error: 'SENSOR_VALUE_TYPE_INT' undeclared (first use in this function)
     attr.type = SENSOR_VALUE_TYPE_INT;
                 ^
/home/neusoft/00_test/zephyr.js/deps/zephyr/scripts/Makefile.build:183: recipe for target 'src/main.o' failed
make[4]: *** [src/main.o] Error 1
/home/neusoft/00_test/zephyr.js/deps/zephyr/Makefile:958: recipe for target 'zephyr-app-dir' failed
make[3]: *** [zephyr-app-dir] Error 2
make[3]: Leaving directory '/home/neusoft/00_test/zephyr.js/arc/outdir/arduino_101_sss'
Makefile:175: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
make[2]: Leaving directory '/home/neusoft/00_test/zephyr.js/deps/zephyr'
/home/neusoft/00_test/zephyr.js/deps/zephyr/Makefile.inc:82: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/neusoft/00_test/zephyr.js/arc'
Makefile:201: recipe for target 'arc' failed
make: *** [arc] Error 2`

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 2345f96 | Jan 9, 2017 | PASS
master | FRDM-K64F  | c707669 | Jan 16,2017 | PASS
master | Arduino 101 | c707669 | Jan 16,2017 | FAIL

jimmy-huang, 2017-02-02:
submitted fix for I2C #669

jimmy-huang, 2017-02-02:
submitted fix for sensor which includes Accelerometer, Gyroscope and AmbientlLight #670

xiaolongx, 2017-02-06:
The I2C and Grove LCD work well with commit 21915c0 on Arduino 101 after #669 merge, and waiting #670 merge to verify sensor related samples

qiaojingx, 2017-02-13:
Sensors work well with commit [3f8eb85](https://github.com/01org/zephyr.js/commits/master) on Arduino 101.

closed 2017-02-13 by qiaojingx