intel/zephyr.js #611: Measure and log boot time of key sample apps

poussa, 2017-01-17:
Our QA is measuring the boot time of some of the sample applications in order to catch improvements and regressions. We should use the performance API to measure and log the boot time of the following sample apps:

samples/HelloWorld.js
samples/I2C.js
samples/WebBLuetoothGrooveLcdDemo.js

The timestamp should be taken when all the modules (via require) have been loaded and all subsystems are initialized. The timestamp should be logged via console.log method.

grgustaf, 2017-01-18:
Should be addressed by #615,

qiaojingx, 2017-01-22:
Boot timestamps has been added to samples/I2C.js, samples/WebBluetoothGroveLcdDemo.js.
A new sample/BootTime.js is added instead of adding boot timestamps to HelloWorld.js.
Verified with commit [5352200](https://github.com/01org/zephyr.js/commit/5352200e9830021aaf670a1722535ac955f5a0dc)

closed 2017-03-31 by grgustaf