intel/zephyr.js #616: Getting Started: Missing step in Mac installation

rwaldron, 2017-01-18:
RE: https://www.zephyrproject.org/doc/getting_started/installation_mac.html

Everything works exactly right until this step: 

```
 cp ${ZEPHYR_BASE}/scripts/cross_compiler/x86.config .config
```

Assuming `ZEPHYR_BASE` refers to the path of a local clone of the git repository, there is no such directory `cross_compiler`: 

```
rwaldron in ~/clonez/zephyr.js/scripts on master
$ tree
.
├── README
├── analyze.sh
├── bldetect
├── checkcopyrights
├── checkheaders
├── convert.sh
├── jsrunner
├── memorystats
├── trlite
└── zjsmaxheap

0 directories, 10 files
```

poussa, 2017-01-19:
Seems that you don't have the ZEPHYR_BASE env variable set. 

You can do it from the zephyr.js repo root directory:

https://github.com/01org/zephyr.js#set-up-zephyr-os-environment-variables

Also remember to source the other settings as in:

https://github.com/01org/zephyr.js#set-up-zjs-environment-variables

rwaldron, 2017-01-19:
Ok, that unblocked the issue—thanks. After that step, I completed the remaining steps for [Installation Mac](https://www.zephyrproject.org/doc/getting_started/installation_mac.html). When I reach the end of that page, I return to [Getting Started](https://www.zephyrproject.org/doc/getting_started/getting_started.html). Under **Building a Sample Application**, the first step _appears_ to instruct to change the value of one env var that was just set on the previous page: 

> Make sure your environment is setup by exporting the following environment variables. When using the Zephyr SDK on Linux for example, type:
>
> `$ export ZEPHYR_GCC_VARIANT=zephyr`

Which conflicts with the "Installation Mac" page: 

> To use the toolchain with Zephyr, export the following environment variables and use the target location where the toolchain was installed, type:
>
> `$ export ZEPHYR_GCC_VARIANT=xtools`

Additionally, the next line creates a new set of problems: 

> ` export ZEPHYR_SDK_INSTALL_DIR=<sdk installation directory>`

There is no mention of the SDK on the "Installation Mac" page. When I go to the downloads page, I can find a downloadable SDK, but I'm not sure that's the correct next step because it doesn't appear on the "Getting Started" page.

poussa, 2017-01-20:
You should have something like this:
```
ZEPHYR_SDK_INSTALL_DIR=/Volumes/CrossToolNG/x-tools
ZEPHYR_GCC_VARIANT=xtools
````

You need to mount the image file by using a script like this:

````
#!/bin/bash
ImageName=CrossToolNG
ImageNameExt=${ImageName}.sparseimage
diskutil umount force /Volumes/${ImageName} && true
hdiutil mount ${ImageNameExt}
cd /Volumes/$ImageName
````

Before you can do the above, you need to create the image first which contains the toolchain. And you need to build the toolchain yourself. The Zephyr SDK (0.8.2) does not contain pre-build binaries for OSX. How to build it is here:

https://www.zephyrproject.org/doc/getting_started/installation_mac.html

At least that is how I have done it. There is new Zephyr SDK (0.9) coming which may have OSX support. I have not checked yet. The `brew install` may install crosstool-ng binaries. I have not checked that either (since I am not using `brew` but `ports`).

I can do more digging next week.

rwaldron, 2017-01-20:
> You should have something like this:

...

> You need to mount the image file by using a script like this:

Yes, and yes: 

```
$ cat .zephyrrc
export XTOOLS_TOOLCHAIN_PATH=/Volumes/CrossToolNG/x-tools
export ZEPHYR_GCC_VARIANT=xtools
rwaldron in ~
$ cat scripts/zephyr-image-mount.sh
#!/bin/bash
ImageName=CrossToolNG
ImageNameExt=${ImageName}.sparseimage
diskutil umount force /Volumes/${ImageName} && true
hdiutil mount ${ImageNameExt}
cd /Volumes/$ImageName
```

(.zephyrrc is sourced in my personal rc which is sourced in .bashrc)

> Before you can do the above, you need to create the image first which contains the toolchain.

As far as I know, I did all of that, unless I'm completely misunderstood? Here's the present state of /Volumes/CrossToolNG: 

```
rwaldron in /Volumes/CrossToolNG
$ ls
.
├── [drwxr-xr-x rwaldron staff     204 Jan 19 14:37]  .build/
├── [drwx------ rwaldron staff    1.2K Jan 19 15:00]  .fseventsd/
├── [drwxr-xr-x rwaldron staff      68 Jan 18 15:11]  build/
├── [lrwxr-xr-x rwaldron staff      70 Jan 19 14:34]  config -> /usr/local/Cellar/crosstool-ng/1.22.0_1/lib/crosstool-ng-1.22.0/config/
├── [drwxr-xr-x rwaldron staff     442 Jan 19 14:34]  config.gen/
├── [drwxr-xr-x rwaldron staff     102 Jan 19 14:37]  x-tools/
├── [-rw-r--r-- rwaldron staff     10K Jan 19 14:37]  .config
├── [-rw-r--r-- rwaldron staff     10K Jan 19 14:37]  .config.2
└── [-rw-r--r-- rwaldron staff     25M Jan 19 15:00]  build.log

6 directories, 3 files
```

What should I verify?

poussa, 2017-01-24:
Everything seems to be in order. For Mac, you should keep these settings:
```
ZEPHYR_SDK_INSTALL_DIR=/Volumes/CrossToolNG/x-tools
ZEPHYR_GCC_VARIANT=xtools
```

Next thing is to type `make` in the zephyr.js root dir. However, we have introduced new regression for Mac and that does not work now. We'll fix that asap. At least I get:

````
[spoussa@zephyr.js (master *)]$ make
Using module: Console
ccflags-y += -DBUILD_MODULE_CONSOLE
sed: 1: "src/Makefile": unterminated substitute pattern
make: *** [analyze] Error 1

rwaldron, 2017-03-31:
Hi! Sorry for the delay, I was on parental leave from January 20th until March 20th and I'm still trying to catch up with everything. I appreciate your patience...

I've followed everything again, from scratch (I nuked every single thing associated with this build). One thing missing from your steps, which should be added in addition to this: 

```
CT_WANTS_STATIC_LINK=n
CT_CC_STATIC_LIBSTDCXX=n
```

Should also have: 

```
CT_CC_GCC_STATIC_LIBSTDCXX=n
```

Otherwise the build ends promptly in error.

Once that was built, I moved back to zephyr.js directory and ran `make`, here's the result:

```
$ make zephyr JS=samples/PWM.js
makecmd: zephyr
Building for Arduino 101...
  RAM allocation: 55KB for X86, 24KB for ARC
  ROM allocation: 256KB for X86, 40KB for ARC
cat: /sdk_version: No such file or directory
Using module: PWM
Using module: A101 Pins
Using module: Console
ccflags-y += -DBUILD_MODULE_PWM -DBUILD_MODULE_A101 -DBUILD_MODULE_TIMER -DBUILD_MODULE_CONSOLE
sed: 1: "src/Makefile": unterminated substitute pattern
make: *** [analyze] Error 1
```

jimmy-huang, 2017-04-01:
sed -i will require an additional argument, so replace sed -i with sed -i ".b"

jimmy-huang, 2017-04-04:
@poussa @rwaldron you can try out my branch here, I am able to build Zephyr with BOARD=qemu_x86, and run hello world, but I am getting a linker error when building the Arduino101 using crosstool-ng compiler.  You'll have to follow the instructions I updated on the README to see if you see the same error.

https://github.com/jimmy-huang/zephyr.js/tree/mac-build-travis

rwaldron, 2017-04-04:
@jimmy-huang thanks for this. I pulled the branch and confirmed that all submodules were up-to-date and the zjs-env.sh. Then I followed the instructions under [Building and running on Mac OSX](https://github.com/jimmy-huang/zephyr.js/tree/mac-build-travis#building-and-running-on-mac-osx). Here are the results of following the updated readme: 

- [x] OSX: Sierra 10.12 or later
  - macOS Sierra 10.12.3
- [x] XCode: 8.1 or later (for make and clang)
  - XCode Version 8.3 (8E162)
- [x] Python: 2.7 or later
  - Python 2.7.10
- [x] Homebrew
  - Homebrew 1.1.11
- [x] Crosstool-NG (if you need to build Zephyr X86 target)
  - Installed per Zephyr docs instructions
- [x] ARC cross-ompiler (for building Arduino 101)
  - Installed per README instructions

- [x] Configure the crosstool-ng
  - Configured per Zephyr docs instructions
- [x] Edit the generated .config, find and replace with these settings
  - Edited per Zephyr docs instructions
- [x] Build crosstool-ng
  - Built per Zephyr docs instructions
- [x] set some environment variables...
```
rwaldron in ~/clonez/zephyr.js on mac-build-travis
$ echo $ZEPHYR_GCC_VARIANT
xtools
rwaldron in ~/clonez/zephyr.js on mac-build-travis
$ echo $XTOOLS_TOOLCHAIN_PATH
/Volumes/CrossToolNG/x-tools
```

Then...

> `make JS=samples/HelloWorld.js`

Which fails on errors concerning unknown flag `-fno-reorder-functions`. After manually removing this, everything appears to work nicely, until the very end:

```
[ 98%] Building C object jerry-core/CMakeFiles/jerry-core.dir/vm/vm.c.obj
clang: warning: optimization flag '-fno-defer-pop' is not supported [-Wignored-optimization-argument]
warning: unknown warning option '-Wlogical-op'; did you mean
      '-Wlong-long'? [-Wunknown-warning-option]
warning: unknown warning option '-Werror=logical-op'; did you mean
      '-Werror=long-long'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-unused-but-set-variable'; did you
      mean '-Wno-unused-const-variable'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-unused-but-set-variable'; did you
      mean '-Wno-unused-const-variable'? [-Wunknown-warning-option]
warning: unknown warning option '-Wno-old-style-declaration'; did you mean
      '-Wno-out-of-line-declaration'? [-Wunknown-warning-option]
warning: unknown warning option '-Werror=format='
      [-Wunknown-warning-option]
/Users/rwaldron/clonez/zephyr.js/deps/jerryscript/jerry-core/vm/vm.c:67:55: error:
      comparison of integers of different signs: 'ecma_integer_value_t'
      (aka 'int') and 'unsigned int' [-Werror,-Wsign-compare]
      bool limit_check = (int_value >= 0 && int_value < (UINT16_MAX + 1));
                                            ~~~~~~~~~ ^  ~~~~~~~~~~~~~~
6 warnings and 1 error generated.
make[5]: *** [jerry-core/CMakeFiles/jerry-core.dir/vm/vm.c.obj] Error 1
make[4]: *** [jerry-core/CMakeFiles/jerry-core.dir/all] Error 2
make[3]: *** [jerry-core/CMakeFiles/jerry-core.dir/rule] Error 2
make[2]: *** [jerry-core] Error 2
make[1]: *** [build/arduino_101/obj-arduino_101/lib/libjerry-core.a] Error 2
make: *** [jerryscript] Error 2
```

jimmy-huang, 2017-04-04:
@rwaldron what version of clang do you have?
Mine is Apple LLVM version 8.1.0 (clang-802.0.38)

Can you also try do make pristine and then rebuild?

Btw, I submitted pr here #968, feel feel to comment on it and we can review them together

jimmy-huang, 2017-04-04:
Btw if you build for Arduino, you will hit a linker error, I filed a bug upstream on Zephyr here
https://jira.zephyrproject.org/browse/ZEP-1994

rwaldron, 2017-04-05:
@jimmy-huang 

> what version of clang do you have?

```
$ clang --version
Apple LLVM version 8.1.0 (clang-802.0.38)
Target: x86_64-apple-darwin16.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

> Can you also try do make pristine and then rebuild?

Done, but ends at the very same place. 

> Btw, I submitted pr here #968, if you check out that branch, it should already have -fno-reorder-functions removed, so I think if you just do make pristine, which rebuilds the snapshot tool, you should not see this err, feel free to comment on it and we can review them together.

I did this separately from your previous instruction, so I've stashed all of my own changes, pulled from your branch (my local is tracking yours), confirmed that I have the latest commits: 

```
$ git log -n 1
commit 84e1f82be78bf97adf01895ab6715318f852e1a2
Author: Jimmy Huang <jimmy.huang@intel.com>
Date:   Wed Apr 5 10:59:20 2017 -0700

    Fixed more warnings

    Signed-off-by: Jimmy Huang <jimmy.huang@intel.com>
```

Then `make pristine`, followed by `make JS=samples/HelloWorld.js` and the result is the same as above.

rwaldron, 2017-04-05:
> and the result is the same as above.

Which is to say that at least most of the errors have been resolved ;)

jimmy-huang, 2017-04-05:
do you still see this? clang: warning: optimization flag '-fno-defer-pop' is not supported [-Wignored-optimization-argument], at least on my machine, i don't see those warnings anymore, it seems like your compiler doesn't understand these no warning flags passed in, maybe can you try edit the deps/jerryscript/targets/zephyr/Makefile.zephyr file, and add a -Wno-sign-compare?

I grepped at the source, maybe you can also try remove or comment out in the deps/zephyr/scripts/Makefile.extrawarn file this line:

KBUILD_CFLAGS += $(call cc-disable-warning, sign-compare)

rwaldron, 2017-04-06:
> do you still see this? clang: warning: optimization flag '-fno-defer-pop' is not supported [-Wignored-optimization-argument], at least on my machine, i don't see those warnings anymore

Correct, these warnings are no longer appearing

> maybe can you try edit the deps/jerryscript/targets/zephyr/Makefile.zephyr file, and add a -Wno-sign-compare?

That gets us to 100% compilation:

```
...
[100%] Linking C static library ../lib/libjerry-core.a
...
```

But then it fails with two errors:

```
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/arch.h:36:
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/arcv2_irq_unit.h:56:2: error: use of unknown builtin
      '__builtin_arc_sr' [-Wimplicit-function-declaration]
        _arc_v2_aux_reg_write(_ARC_V2_IRQ_SELECT, irq);
        ^
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/aux_regs.h:117:41: note: expanded from macro
      '_arc_v2_aux_reg_write'
#define _arc_v2_aux_reg_write(reg, val) __builtin_arc_sr((unsigned int)val, (volatile uint32_t)reg)
                                        ^
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/arch/arc/core/offsets/offsets.c:26:
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/kernel/include/kernel_structs.h:10:
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/kernel.h:3416:
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/cpu.h:17:
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/arch.h:37:
In file included from /Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/asm_inline.h:18:
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/asm_inline_gcc.h:41:19: error: use of unknown builtin
      '__builtin_arc_lr' [-Wimplicit-function-declaration]
        return (uint8_t)(_arc_v2_aux_reg_read(port) & 0x000000ff);
                         ^
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/aux_regs.h:116:35: note: expanded from macro
      '_arc_v2_aux_reg_read'
#define _arc_v2_aux_reg_read(reg) __builtin_arc_lr((volatile uint32_t)reg)
                                  ^
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/asm_inline_gcc.h:41:19: note: did you mean
      '__builtin_arc_sr'?
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/aux_regs.h:116:35: note: expanded from macro
      '_arc_v2_aux_reg_read'
#define _arc_v2_aux_reg_read(reg) __builtin_arc_lr((volatile uint32_t)reg)
                                  ^
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/arcv2_irq_unit.h:56:2: note: '__builtin_arc_sr'
      declared here
        _arc_v2_aux_reg_write(_ARC_V2_IRQ_SELECT, irq);
        ^
/Users/rwaldron/clonez/zephyr.js/deps/zephyr/include/arch/arc/v2/aux_regs.h:117:41: note: expanded from macro
      '_arc_v2_aux_reg_write'
#define _arc_v2_aux_reg_write(reg, val) __builtin_arc_sr((unsigned int)val, (volatile uint32_t)reg)
                                        ^
````

jimmy-huang, 2017-04-06:
It looks like you fail to build the ARC image, what command did you use to build? like

make JS=samples/HelloWorld.js ARC_CROSS_COMPILE=/Volumes/CrossToolNG/arc-elf32/1.6.9+1.0.1/bin/arc-elf32-

Maybe if you can post the full build ouput on pastebin?

rwaldron, 2017-04-06:
https://pastebin.com/03DgTSi2

jimmy-huang, 2017-04-07:
@rwaldron ok, i diffed the build log with yours and mine, and it seems that you environment isn't building with crosstool-ng cross compiler, no wonder you are seeing all kinds of compiler warnings:  see this line on your build output, you are compiling with apple's clang:

-- Found assembler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang

On my system it builds with the crosstool-ng 

-- Found assembler: /Volumes/CrossToolNG/x-tools/i586-pc-elf/bin/i586-pc-elf-gcc

You need to figure out what environment variables you are setting, you should see something like this when you do export:

declare -x XTOOLS_TOOLCHAIN_PATH="/Volumes/CrossToolNG/x-tools"
declare -x ZEPHYR_BASE="/Users/zhuang1/projects/zephyrjs/zephyr.js/deps/zephyr"
declare -x ZEPHYR_GCC_VARIANT="xtools"
declare -x ZJS_BASE="/Users/zhuang1/projects/zephyrjs/zephyr.js"

Also we fixed a few bugs on master on some of the jerryscript flags, and we merged the osx patch, so you can do a git checkout of master to see the latest fixes.

rwaldron, 2017-04-07:
>  diffed the build log with yours and mine, and it seems that you environment isn't building with crosstool-ng cross compiler

I feel so silly... I forgot to rerun the script to mount /Volumes/CrossToolNG ... *facepalm*

On latest master, after running scripts/osxmountzephyr.sh, I walked through building crosstool-ng from scratch. Reran `source  ...` for all the env files and finally...

```
To exit from QEMU enter: 'CTRL+a, x'
[QEMU] CPU: qemu32
qemu-system-i386: warning: Unknown firmware file in legacy mode: genroms/multiboot.bin

Hello, ZJS world!
```

rwaldron, 2017-04-07:
I'm still unable to build a zephyr.bin.dfu to push out to the board—is that expected?

jimmy-huang, 2017-04-07:
Yes, what board do you have? Arduino 101? If so, yeah, there's an issue right now which I filed the bug against, I believe if you have a frdm-k64f, then with the ARM compiler, you can flash it, but I didn't try that, yet.

rwaldron, 2017-04-07:
I do have a frdm-k64f, but it's at the office, so I'll have to pick it up later or wait until monday. I can now test the require/modules PR though, so that's awesome :)

jimmy-huang, 2017-04-07:
@rwaldron that's good, I am sure you might run into issues later, but feel free to submit an issue or PR if you see fit so we can update the documentation as well on latest instructions building on Mac

rwaldron, 2017-04-07:
> I can now test the require/modules PR though

Sorry, this is actually not true. I pulled that branch and rebased it with master, but the build fails.

jimmy-huang, 2017-04-08:
@rwaldron what pr?  maybe I can try to test it.

jimmy-huang, 2018-01-25:
Closing this as this has been fixed by #1780

closed 2018-01-25 by jimmy-huang