intel/zephyr.js #620: [console] Not support to console.assert()

cuiyanx, 2017-01-19:
Set parameter as "true", console.assert() is not work; Set parameter as "false", console.assert() throw out error.

jprestwo, 2017-01-19:
The correct usage for console.assert() is:

```
console.assert(value[, message][, ...args])
```

Specifying a string as the first parameter won't work. What is strange is that you are seeing the "string is too long" print. When I specify a string e.g. "true" as the first parameter I get:
```
[ERROR] invalid parameters
```
If you use a true boolean value as the first parameter it behaves normally. Using a false boolean value also behaves correctly by throwing an error. Are you seeing behavior different than this?

jprestwo, 2017-01-19:
Sorry I was not understanding how you produced these results. I see the problem your seeing now when you specify a message in the assert call.

cuiyanx, 2017-02-14:
fixed and closed

closed 2017-02-14 by cuiyanx