intel/zephyr.js #621: [console] Not support placeholders for Console APIs

cuiyanx, 2017-01-19:
Set "%d" for number and "%s" for string, there are not work.

Code : 
```
var arg1 = 1024;
console.log("console.log: %d", arg1);
var arg2 = "My love!";
console.log("console.log: %s", arg2);
```

grgustaf, 2017-01-19:
I don't think it makes sense for us to implement support for this, unless it's already present in JerryScript or something and we can reuse it. So probably @jprestwo you can resolve this by specifying in the documentation that format specifiers are not supported. This was a surprise to me, I never knew this existed. :) But it's there in both Node and e.g. Chrome.

jprestwo, 2017-03-02:
Docs added with: https://github.com/01org/zephyr.js/pull/787

cuiyanx, 2017-03-03:
OK，thanks and closed.

closed 2017-03-03 by cuiyanx