intel/zephyr.js #622: [console] Not support constructor to create

cuiyanx, 2017-01-19:
Node.js v7.4.0
`Creates a new Console by passing one or two writable stream instances.`

When I create constructor for my console class, I am failed.

grgustaf, 2017-01-19:
FYI we aim to support a subset of the LTS version of Node.js APIs (currently 6.9.4) vs. the latest and greatest (currently 7.4.0). But that doesn't change this item.

I think this bug is really just talking about some basic constructor problem? But it does point out that eventually we might want users to be able to create consoles that log to files on the flash filesystem, and/or UART or something.

jprestwo, 2017-02-27:
@grgustaf, @cuiyanx,

I think for now we don't need a console constructor because we only support printing to "stdout" (UART), but we could change this to an enhancement/feature request for allowing a constructor to be called on a stream e.g. UART1/2, FS etc.

jprestwo, 2017-03-22:
@grgustaf, @cuiyanx,

Can we close this and reopen a feature request instead.

cuiyanx, 2017-03-23:
@jprestwo OK, thanks for your attention。

closed 2017-03-23 by cuiyanx