intel/zephyr.js #626: "make linux" always rebuilds from scratch

pfalcon, 2017-01-19:
E.g.:

~~~
$ time make linux
[skip complete or almost complete rebuild]
real	0m21.232s
user	0m14.746s
sys	0m1.562s
$ time make linux
[skip complete or almost complete rebuild]
real	0m21.737s
user	0m14.997s
sys	0m1.463s
~~~

20s is too slow for every minor change ;-).

grgustaf, 2017-02-01:
Fixed by commit 0b5ebae, PR #662, I believe.

xiaolongx, 2017-02-03:
The time reduced to about 10 seconds on commit 6c29671
```
$ time make linux
real    0m10.529s
user    0m9.384s
sys     0m0.508s
$ time make linux
real    0m10.706s
user    0m9.484s
sys     0m0.556s
```

qiaojingx, 2017-02-15:
The time is less than 10 seconds on commit be6e80f.
```
$ time make BOARD=linux
real	0m8.649s
user	0m7.980s
sys	0m0.376s
$ time make BOARD=linux
real	0m8.980s
user	0m8.280s
sys	0m0.320s
```

pfalcon, 2017-02-15:
Yes, I believe this is fixed now, closing.

closed 2017-02-15 by pfalcon