intel/zephyr.js #638: [Linux][Buffer] Unable to run buffer test on linux

xiaolongx, 2017-01-22:
Unable to run [Buffer.js](https://github.com/01org/zephyr.js/blob/master/samples/Buffer.js), [BufferMemory.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/BufferMemory.js), [test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js), [test-buffer-rw.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer-rw.js) using below commands on linux, but it should be supported by #634 merged. 
```
/home/ZJS/ww04/zephyr.js$ make linux
/home/ZJS/ww04/zephyr.js$ ./outdir/linux/release/jslinux samples/Buffer.js
```
Output:
```
Buffer test...

src/main.c:217 main():
[ERROR] JerryScript: cannot run javascript
```

Branch | Target Device | Commid ID | Result
---|---|---|---
master | Arduino 101 | 5352200 | PASS
master | FRDM-K64F | 5352200 | PASS
master | ubuntu 16.04 LTS 64-bit | 5352200 | FAIL

jprestwo, 2017-01-23:
@grgustaf, we found this out the other day. Just a compiler problem. Did you add this to that change you had? If not I can push a fix really quick.

jprestwo, 2017-01-23:
Minor fix with https://github.com/01org/zephyr.js/pull/644

xiaolongx, 2017-02-03:
This issue cannot reproduce with commit 6c29671, close

closed 2017-02-03 by xiaolongx