intel/zephyr.js #639: [error] Not instance of errorType

xiaolongx, 2017-01-22:
Flashing [test-error.js](https://github.com/01org/zephyr.js/blob/master/tests/test-error.js) on both Arduino 101 and FRDM-K64F with commit 5352200, 3 test case of `expectStandardError` were failed.

Error logs:
```
Testing error types...

PASS - SecurityError works as expected
PASS - NotSupportedError works as expected
PASS - SyntaxError works as expected
PASS - TypeError works as expected
PASS - RangeError works as expected
PASS - TimeoutError works as expected
PASS - NetworkError works as expected
PASS - SystemError works as expected
calling require with an integer: expected error object
FAIL - calling require with an integer
calling require with a long name: expected error object
FAIL - calling require with a long name
calling require with an invalid module name: expected error object
FAIL - calling require with an invalid module name
TOTAL: 8 of 11 passed
```

grgustaf, 2017-02-01:
I suspect this was fixed with commit 7c9fbe5, a few commits later.

xiaolongx, 2017-02-03:
This issue cannot reproduce on both Arduino 101 and FRDM-k64F with commit 6c29671, close

closed 2017-02-03 by xiaolongx