intel/zephyr.js #640: [Linux][event] The "default number of listener" is not equal to 0

xiaolongx, 2017-01-22:
Run [test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js) on linux via below commands, the "default number of listeners" exist extra one that works.
```
/home/ZJS/ww04/zephyr.js$ make linux
/home/ZJS/ww04/zephyr.js$ ./outdir/linux/release/jslinux tests/test-event.js
```
Output:
`FAIL - event: default number of listeners`

Branch | Target Device | Commit ID | Result
---|---|---|---
master | Arduino 101 | 5352200 | PASS
master | FRDM-K64F | 5352200 | PASS
master | ubuntu 16.04 64-bit | 5352200 | FAIL

jprestwo, 2017-01-23:
This should be fixed with this PR: https://github.com/01org/zephyr.js/pull/645

I did have to modify the tests because it assumed that the default number of max listeners was 0, which was not true. There is now a object property in the event emitter which is `EventEmitter.defaultMaxListeners`. This starts as 10 (which is arbitrary, Node does not specify a default),  but the actual value can be changed with `EventEmitter.setMaxListeners()`. Calling `setMaxListeners()` will not change this default value though. The change I made was to compare the return from `setMaxListeners()` to the new object property `defaultMaxListeners`.

xiaolongx, 2017-02-03:
this issue cannot reproduce with commit 6c29671, close

closed 2017-02-03 by xiaolongx