intel/zephyr.js #643: zjs_callbacks: "Multiply signalled callback called only once" paradigm - does it really work?

pfalcon, 2017-01-22:
https://github.com/01org/zephyr.js/blob/master/src/zjs_callbacks.h#L155 says:

> You can signal callbacks multiple times, but if the callback has not been serviced between signaling, it will only get called once.

However, I cannot identify code which would ensure "will only get called once". zjs_signal_callback() just adds a call func id/arguments to ringbuffer, and neither Zephyr docs on ringbuffer nor Linux implementation in Z.js appear to mention/have any "duplicate filtering".

Note that in my case, "duplicate filtering" is not desirable, and I actually work it around by not pre-registering a callback, but queuing it on spot using zjs_add_callback_once/ zjs_signal_callback (and thus never storing id). But then I got curious how that "duplicate filtering" implemented, and well, it's not.

I'd actually argue that "no duplicate filtering" is the behavior desired in most cases. Consider for example "data received callback" for UART, I2C, socket, whatever - any data received should be queued up for delivery to a callback, losses are not acceptable. "Duplicates filtering" behavior may be useful in peculiar cases. For example, if a user attaches a callback to a fast-changing GPIO, it may have adverse effects like DoS system heap by uncontrollable cb queue growth or DoS other callbacks and system performance overall.

So, it would be nice to get this matter cleared - if duplicate filtering is supported, it would be nice to have documented way of getting "lossless" callbacks. If it's not supported, perhaps clauses related to it should be removed from docstrings, and at later time, new APIs should be introduced to deal with it.

jprestwo, 2017-01-23:
@pfalcon:

Good catch. This comment never got changed after we added the ring buffer as the callback queue mechanism. Before, we just had a static list and set an int to signal a callback, which meant it would overwrite and previous signaled callbacks if they had not been serviced (for any given ID). This also meant the callback arguments would get overwritten. Now, with the ring buffer, we queue up the callback function and the callback arguments which means you can signal the same callback many times before it is serviced and it will always get called as many times as its signaled with a unique set of arguments.

@grgustaf, you can assign this to me to update the comment.

grgustaf, 2017-01-26:
I meant to talk to @jprestwo about that comment when I saw a it a few days ago, because yeah part of the point of the new ring buffer implementation is that we do remember *all* pending callbacks. So the comment is just incorrect unless I'm very much mistaken.

jprestwo, 2017-03-02:
@grgustaf, can we close this?

jprestwo, 2017-03-22:
@grgustaf, @pfalcon,

Can we close this?

grgustaf, 2017-03-29:
For quite a while, the ring buffer has been there and "lossless" duplicate callbacks were supported.

One more refinement: As of commit 1825095, when a callback is removed, a "flush" command is inserted into the ring buffer and the callback ID may not be reassigned until that flush makes it through. So for the case of the "once" callbacks this does guarantee the callback will only be called once.

Anyway, long since solved I believe, closing.

closed 2017-03-29 by grgustaf