intel/zephyr.js #651: compile warnings when building samples/I2C.js

grgustaf, 2017-01-24:
After the recent reorganization to I2C there are a bunch of compile warnings when I do:

`$ make JS=samples/I2C.js`

Please fix those and make sure we're doing the right thing with error handling. I started to look at it myself and thought I'd better leave it to the expert.

brianjjones, 2017-01-27:
@grgustaf The fix can be found in PR https://github.com/01org/zephyr.js/pull/663

brianjjones, 2017-01-31:
@grgustaf this issue has been fixed and can be closed.

grgustaf, 2017-01-31:
@brianjjones, if I do `make clean; make JS=samples/I2C.js`, I still see two compile warnings:

```
  CC      src/main.o
/home/pepsi/dev/zjs/project/src/main.c: In function 'stop_js_handler':
/home/pepsi/dev/zjs/project/src/main.c:77:5: warning: implicit declaration of function 'zjs_ipm_free_callbacks' [-Wimplicit-function-declaration]
     zjs_ipm_free_callbacks();

...

  CC      src/zjs_i2c.o
/home/pepsi/dev/zjs/project/src/zjs_i2c.c: In function 'zjs_i2c_init':
/home/pepsi/dev/zjs/project/src/zjs_i2c.c:249:5: warning: implicit declaration of function 'zjs_i2c_ipm_init' [-Wimplicit-function-declaration]
     zjs_i2c_ipm_init();
     ^
```

brianjjones, 2017-01-31:
@grgustaf That's due to to my stop_js changes, nothing to do with I2C.  Fix is ready, see it here -> https://github.com/01org/zephyr.js/pull/666

grgustaf, 2017-01-31:
@brianjjones yeah, but the problem is that when I build I2C.js there are warnings. It doesn't matter if they have to do with I2C. :)

Do you see that second one about i2c init? That's still there with your new patch, and *does* have to do with I2C. :)

xiaolongx, 2017-02-06:
No warnings output after executing commands: `make clean; make JS=samples/I2C.js` with commit 21915c0.

grgustaf, 2017-03-01:
Verified, closing.

closed 2017-03-01 by grgustaf