intel/zephyr.js #656: JS Callback argument management is cumbersome

pfalcon, 2017-01-26:
How JerryScript works, is that any jeryr_value_t must be released once its use if finished. That means that to execute any callback which accepts args one must: 1) store all args in an adhocly allocated chunk of memory; 2) have a post function which is passed a pointer to this chunk, which will go thru it to release values stored in it and free chunk itself.

At the same time, callback args are already stored in callback queue ring buffer. And if a callback is a JS callback, we know that stuff stored in the ring buffer is an array of jerry_value_t's. Then, zjs_service_callbacks(), which fetches these args from the ring buffer, is the best place to release them. For niche cases when value of an argument of a callback should not be released upon return, a caller can use jerry_acquire_value() on it.

grgustaf, 2017-01-26:
Yeah, I just went through a huge review of all our code apart from OCF stuff (which we're going to look at today) reviewing for these exact problems. See commit 8d6571a. I did discover (and fix!) the exact issue you're talking about, but it looks like I left that out of the final patch. I suspect it's sitting on my PC at home, so I'll plan to check that tonight.

grgustaf, 2017-01-28:
This should be resolved by #657 and #661.

closed 2017-01-28 by pfalcon