intel/zephyr.js #672: [REG] Fail to build with "SNAPSHOT=on" on zjs-0.2 branch

xiaolongx, 2017-02-03:
The Snapshot build failed with command: $ make JS=samples/AIO.js SNAPSHOT=on all on commit eafdfe9, the issue #588 was fixed after #592 merge, but now no commit #592 on zjs-0.2 branch.

Error logs:
```
artificial>:(.text.main+0x25): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0x77): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0xc9): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0x10f): undefined reference to `zjs_shorten_filepath'
<artificial>:(.text.main+0x14f): undefined reference to `zjs_shorten_filepath'
/tmp/cchFTt6c.ltrans0.ltrans.o:<artificial>:(.text.main+0x1a6): more undefined references to `zjs_shorten_filepath' follow
collect2: error: ld returned 1 exit status
Makefile.snapshot:84: recipe for target 'snapshot' failed
make[1]: *** [snapshot] Error 1
make[1]: Leaving directory '/home/xiaolong/00_xiaolong/ZJS/ww04/zjs-0.2/zephyr.js'
Makefile:178: recipe for target 'generate' failed
make: *** [generate] Error 2

```

haoxli, 2017-02-06:
This issue block zjs-0.2 promote to stable release.

jimmy-huang, 2017-02-08:
patch submitted #698

qiaojingx, 2017-02-10:
This issue is not reproduced on Arduino 101 with commit [326b8dd](https://github.com/01org/zephyr.js/commit/326b8ddf914e5d63a9cb32e25ab32dcb4145b9df) in zjs-0.2 branch.

closed 2017-02-13 by xiaolongx