intel/zephyr.js #674: [GenericSensor] How can we trigger onerror event?

haoxli, 2017-02-04:
Following [W3C spec](https://www.w3.org/TR/2016/WD-generic-sensor-20160830/), onerror should be sent when sensor is not created correctly, or start() call has failed. We try many ways, but only can get error log, the error event still cannot be triggered. 
```
var sensor = new Accelerometer({});

sensor.onstatechange = function(event) {
    console.log("state: " + event);
};

sensor.onerror = function(event) {
    console.log("error: " + event.error.name +
                " - " + event.error.message);
};

sensor.start();
```
Error log:

> src/zjs_sensor.c:429 zjs_sensor_create():
[ERROR] unsupported frequency, default to 20Hz
ipm_console0: 'Sensor core running ZJS ARC support image'
state: activating
state: activated

grgustaf, 2017-02-08:
I'm not sure how it's working currently, but in our discussion with Zoltan Friday, he suggested that where we see something like onstatechange we change it to on('statechange'). We would try to make this explicitly clear in our API docs.

@jimmy-huang should know what the current behavior is.

grgustaf, 2017-02-28:
@jimmy-huang please update with any info on how to trigger?

The W3C samples seem to really use "onerror" etc. not on('error'), so we'll stick with that to match existing coce.

jimmy-huang, 2017-03-03:
@haoxli yeah, it's hard to trigger it on the accelerometer and gyroscope, because unless there's a communication error of some sort in the ipm, we don't report an error, but with the AmbientLightSensor, you should be able to trigger it, for example, when you don't pass the pin object in the constructor, I am currently working on a patch that you can at least show that.

jimmy-huang, 2017-03-14:
@haoxli, you should be able to trigger the error now with the latest code,i.e, if you construct a AmbientLight Sensor without passing a pin, please verify

haoxli, 2017-03-15:
@jimmy-huang thanks for your information. I've verified it PASS on commit 9777997095031a68376728a7f0b2da9342e7fb33 with the code:
```
var sensor = new AmbientLightSensor({
    frequency: 50
});

sensor.onstatechange = function(event) {
    console.log("state: " + event);
};

sensor.onerror = function(event) {
    console.log("error: " + event.error.name + " - " + event.error.message);
};

sensor.start();
```
Logs:
```
Testing ambientlight APIs
state: activating
state: errored
error: DataError - pin not found
```

closed 2017-03-15 by haoxli