intel/zephyr.js #676: [REG] I2C does not work on FRDM-K64F due to "I2C_SS_0 not found"

xiaolongx, 2017-02-06:
I2C does not work on FRDM-K64F, it occurred after #669 was merged which fixed I2C and GroveLCD issue on Arduino 101.

Error Logs:
```
Boot timestamp: 0

I2C sample...

src/zjs_i2c_handler.c:20 zjs_i2c_handle_open():
[ERROR] I2C bus I2C_SS_0 not found.

src/main.c:220 main():
[ERROR] JerryScript: cannot run javascript
```

Branch | Target Device | Commit ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 21915c0 | Feb 6, 2017 | PASS
master | FRDM-K64F | 21915c0 | Feb 6, 2017 | FAIL
master | FRDM-K64F | 10086db | Feb 6, 2017 | PASS

brianjjones, 2017-02-08:
@grgustaf @xiaolongx The fix for this has been merged into master.

grgustaf, 2017-02-08:
This was PR #680 and commit eefc65f. Thanks @brianjjones.

qiaojingx, 2017-02-15:
This issue is not reproduced on FRDM-K64F with commit 3f8eb85.

closed 2017-02-15 by xiaolongx