intel/zephyr.js #677: [REG] Fail to require of JS modules

xiaolongx, 2017-02-06:
Fail to require of JS modules with command `var bmp280 = require("BMP280.js");` on both Arduino 101 and FRDM-K64F, it affect [I2CBMP280.js](https://github.com/01org/zephyr.js/blob/master/samples/I2CBMP280.js) and [test-event.js](https://github.com/01org/zephyr.js/blob/master/tests/test-event.js).

Error Logs:
```
src/main.c:220 main():
[ERROR] JerryScript: cannot run javascript
```

Branch | Target Device | Commit ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 21915c0 | Feb 6, 2017 | FAIL
master | FRDM-K64F | 21915c0 | Feb 6, 2017 | FAIL

jimmy-huang, 2017-02-08:
submitted patch #699

qiaojingx, 2017-02-15:
This issue is not reproduced on Arduino 101 and FRDM-K64F with commit 3f8eb85.

closed 2017-02-15 by xiaolongx