intel/zephyr.js #678: [I2C][REG] Fail to throw error when opening an invalid bus

xiaolongx, 2017-02-06:
Flashing [test-i2c.js](https://github.com/01org/zephyr.js/blob/master/tests/test-i2c.js) with commit 21915c0 on Arduino 101, fail to throw error when opening an invalid bus, but not sure if it exist on FRDM-K64F due to "I2C_SS_0 not found" issue #676.
```
expectThrow("I2C: open invalid bus", function () {
    i2c.open({ bus: 1, speed: 100 });
});
```

Serial Logs:

Grove LCD is required...
```
PASS - I2C: open with speed 10
PASS - I2CBus: bus property exist
PASS - I2CBus: bus is readonly property
PASS - I2CBus: speed property exist
PASS - I2CBus: speed is readonly property
PASS - I2C: open with speed 100
PASS - I2C: open with speed 400
PASS - I2C: open with speed 1000
PASS - I2C: open with speed 34000
PASS - I2C: write() and read()
PASS - I2C: burstRead()
FAIL - I2C: open invalid bus
TOTAL: 11 of 12 passed
```

Branch | Target Device | Commit ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 21915c0 | Feb 6, 2017 | FAIL
master | Arduino 101 | 2345f96 | Jan 5, 2017 | PASS

qiaojingx, 2017-02-14:
This issue is still reproduced on Arduino 101 with commit 3f8eb85. But can not be reproduced on FRDM-K64F.
Serial logs:
```
Grove LCD is required...
PASS - I2C: open with speed 10
PASS - I2CBus: bus property exist
PASS - I2CBus: bus is readonly property
PASS - I2CBus: speed property exist
PASS - I2CBus: speed is readonly property
PASS - I2C: open with speed 100
PASS - I2C: open with speed 400
PASS - I2C: open with speed 1000
PASS - I2C: open with speed 34000
PASS - I2C: write() and read()
PASS - I2C: burstRead()
ipm_console0: ''
ipm_console0: 'src/zjs_i2c_handler.c:26 zjs_i2c_handle_open():'
ipm_console0: '[ERROR] I2C bus I2C_SS_1 not found.'
FAIL - I2C: open invalid bus
TOTAL: 11 of 12 passed
```

brianjjones, 2017-03-21:
The Arduino 101 actually does have a bus 1.  However, I2C_SS_1 isn't connected to anything so it doesn't make sense to allow users to try and access it.  I've added a check specific to A101 to ensure it doesn't get used.  The test above should work with https://github.com/01org/zephyr.js/pull/858

commit a6c321ab (2017-03-22): [i2c] Disallow invalid bus #1 on Arduino 101 (#858)

I2C will now throw an error if you try and access the
I2C_SS_1 bus on Arduino 101, since it's not connected
to anything.

Fixes #678.
closed 2017-03-22 by grgustaf