intel/zephyr.js #679: [REG] Error thrown after finishing test-grovelcd-manual tests

xiaolongx, 2017-02-06:
Error thrown after running [test-grovelcd-ds-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-ds-manual.js), [test-grovelcd-rgb-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-grovelcd-rgb-manual.js) with commit 21915c0 on Arduino 101, but not sure if it exist on FRDM-K64F due to Grove LCD build issue #675.

Error logs:
```
test GroveLCD display
ipm_console0: 'Sensor core running ZJS ARC support image'
setFunction(row, mode, dot): expected result ROW1, 8BIT
setFunction(row, mode, dot): expected result ROW2, 8BIT
setFunction(row, mode, dot): expected result 8BIT, BIG
setFunction(row, mode, dot): expected result 8BIT, LIT
setFunction(row, mode, dot): expected result ROW1, BIG
setFunction(row, mode, dot): expected result ROW1, LIT
setFunction(row, mode, dot): expected result ROW2, BIG
setFunction(row, mode, dot): expected result ROW2, LIT
setFunction(row, mode, dot): expected result ROW1, 8BIT, BIG
setFunction(row, mode, dot): expected result ROW1, 8BIT, LIT
setFunction(row, mode, dot): expected result ROW2, 8BIT, BIG
setFunction(row, mode, dot): expected result ROW2, 8BIT, LIT
Testing completed

src/zjs_callbacks.c:445 print_error_message():
[ERROR] Uncaught exception: Error: Base object cannot be null or undefined.
```
```
test RGB color
ipm_console0: 'Sensor core running ZJS ARC support image'
setColor(R, G, B): expected result 255, 0, 0
color gradient:  red
setColor(R, G, B): expected result 0, 255, 0
color gradient: green
setColor(R, G, B): expected result 0, 0, 255
color gradient:  blue
setColor(R, G, B): expected result 255, 255, 0
color gradient: yellow
setColor(R, G, B): expected result 255, 0, 255
color gradient: purple
setColor(R, G, B): expected result 0, 255, 255
color gradient:  cyan
setColor(R, G, B): expected result 255, 255, 255
color gradient: white
Testing completed

src/zjs_callbacks.c:445 print_error_message():
[ERROR] Uncaught exception: Error: Base object cannot be null or undefined.
```
Branch | Target Device | Commit ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 21915c0 | Feb 6, 2017 | FAIL
master | Arduino 101 | 2345f96 | Jan 5, 2017 | PASS

qiaojingx, 2017-02-15:
This issue is also reproduce on FRDM-K64F with commit 3f8eb85

qiaojingx, 2017-03-27:
Error is still thrown.
[image]
[image]

grgustaf, 2017-03-29:
I can reproduce this too.

jimmy-huang, 2017-03-29:
This appears to be a bug in the tests where after the last test is complete, counter has been incremented to be 7, but there are only 7 elements in the array (0-6), so basically an array out of bound error, but the remainder of the function will continue to run after calling clearInterval() so either add checks to make sure index is inbound or terminate immediately.  I submitted patch #933 to fix this.

qiaojingx, 2017-03-30:
Verified with commit a2ce873 on Arduino 101 and FRDM-k64f. This issue is not reproducible. Closed

closed 2017-03-30 by xiaolongx