intel/zephyr.js #682: [Timers] 'Segmentation fault' output on Linux

xiaolongx, 2017-02-07:
'Segmentation fault' output and then exit automatically when running command: `./outdir/linux/release/jslinux samples/Timers.js` with commit 21915c0, this issue cannot reproduce on both Arduino 101 and FRDM-K64F.

```
Starting Timers example...
Interval (args) #1 arg1: 1 arg2: 2
Interval (no args) #2
Interval (args) #2 arg1: 1 arg2: 2
Interval (no args) #3
Interval (args) #3 arg1: 1 arg2: 2
Interval (no args) #4
Interval (args) #4 arg1: 1 arg2: 2
Interval (no args) #5
Timeout, clearing interval
Segmentation fault (core dumped)

```

qiaojingx, 2017-02-15:
This issue is not reproduced Linux with commit 3f8eb85.

xiaolongx, 2017-02-21:
This issue is reproduced on Linux again with commit 7c5ff3c.
Reopen.

grgustaf, 2017-03-02:
Should be fixed by #780.

Also, the output of this test will change with #781. Note the #1/#2 output is strange: 1-2-2-3-3-4-5. It will be 1-1-2-2-3-3-4-4-5-5 now, which is what happens on Chrome and probably what the author intended.

closed 2017-03-02 by jimmy-huang

qiaojingx, 2017-03-06:
Verified with commit b29dbbc, this issue is not reproducible when running Timers.js on Linux.
But 'Segmentation fault' issue occurred when running test-timers.js and ocfServer.js on Linux.
Create a new issue #792.

commit 18250957 (2017-03-02): [callbacks] Retain removed callbacks until pending callbacks flushed

When a callback is removed, it may already have been signaled, which
means items are waiting in the ring buffer to be executed. If the same
callback ID were reassigned in the meantime, that pending callback
could execute the wrong code with the wrong arguments. So instead keep
the callback around, marked for removal, add a special flush command
into the ring buffer, and once it comes through we know it's safe to
actually remove the callback.

Also, reorder items in zjs_callback_t struct for better packing.

Fixes #682.