intel/zephyr.js #683: [Buffer] The invalid data processing on Linux is inconsistent with Arduino 101 and FRDM-K64F

xiaolongx, 2017-02-07:
The invalid data processing on Linux is inconsistent with Arduino 101 and FRDM-K64F when running [test-buffer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-buffer.js) via command `zephyr.js$ ./outdir/linux/release/jslinux tests/test-buffer.js` with commit 21915c0

Error Logs:
```
test-buffer.js:
FAIL - The length of Buffer(-1) expected:0 got:4294967295
PASS - The length of Buffer(0) expected:0 got:0
PASS - The length of Buffer(8) expected:8 got:8
PASS - The length of Buffer(8.5) expected:8 got:8
PASS - The length of Buffer(4294967295) expected:4294967295 got:4294967295
FAIL - The length of Buffer(4294967296) expected:4294967295 got:0
PASS - The length of Buffer is readonly
FAIL - The value of readUInt8() expected:0 got:255 when writeUInt8(-1)
PASS - The value of readUInt8() expected:0 got:0 when writeUInt8(0)
PASS - The value of readUInt8() expected:1 got:1 when writeUInt8(1)
PASS - The value of readUInt8() expected:1 got:1 when writeUInt8(1.6)
PASS - The value of readUInt8() expected:255 got:255 when writeUInt8(255)
PASS - The value of readUInt8() expected:0 got:0 when writeUInt8(256)
PASS - The value of readUInt8() expected:44 got:44 when writeUInt8(300)
PASS - Error thrown when the offset of writeUInt8() is outside the bounds of the Buffer
PASS - Error thrown when the offset of readUInt8() is outside the bounds of the Buffer
PASS - The value of toString('hex') expected:0011ff00ff got:0011ff00ff
PASS - Error thrown when 'hex' is not given to toString()
TOTAL: 15 of 18 passed
```

qiaojingx, 2017-03-27:
Test cases have been adjusted.
Error logs as below(commit bb11854):
```
FAIL - The length of Buffer(-1) expected:0 got:4294967295
PASS - The length of Buffer(0) expected:0 got:0
PASS - The length of Buffer(8) expected:8 got:8
PASS - The length of Buffer(8.5) expected:8 got:8
PASS - The length of Buffer(4294967295) expected:4294967295 got:4294967295
FAIL - The length of Buffer(4294967296) expected:4294967295 got:0
PASS - The length of Buffer is readonly
PASS - The value of readUInt8() expected:0 got:0 when writeUInt8(0)
PASS - The value of readUInt8() expected:1 got:1 when writeUInt8(1)
PASS - The value of readUInt8() expected:1 got:1 when writeUInt8(1.6)
PASS - The value of readUInt8() expected:255 got:255 when writeUInt8(255)
PASS - The value of readUInt8() expected:0 got:0 when writeUInt8(256)
PASS - The value of readUInt8() expected:44 got:44 when writeUInt8(300)
PASS - Error thrown when the offset of writeUInt8() is outside the bounds of the Buffer
PASS - Error thrown when the offset of readUInt8() is outside the bounds of the Buffer
PASS - The value of toString('hex') expected:0011ff00ff got:0011ff00ff
PASS - Error thrown when 'hex' is not given to toString()
PASS - The value of readUInt32BE() with precision 3 expected: -0.232 got: -0.232
TOTAL: 16 of 18 passed
```

grgustaf, 2017-03-29:
Results are now consistent between A101 and FRDM-K64F. As per comments in #916, the Buffer constructor has changed a bit and the test should be updated.

qiaojingx, 2017-03-31:
Still got two inconsistent results on Linux.
1.  buffer.kMaxLength got 2^32-1 on Linux now, but it should be 2^31-1 according to [buffer_buffer_kmaxlength](https://nodejs.org/dist/latest-v6.x/docs/api/buffer.html#buffer_buffer_kmaxlength).
2. No error thrown, but got 0, when it overflow.

Please refer to the following pic.
[image]

qiaojingx, 2017-04-06:
Will verify this issue after buffer blocking issue #970 fixed.

qiaojingx, 2017-04-07:
Verified with commit f8f9621 on Linux. buffer.kMaxLength got 2^31-1. And error thrown when it overflow.
[image]

CLosed as fixed.

closed 2017-04-07 by xiaolongx