intel/zephyr.js #685: DEBUG build is broken

poussa, 2017-02-07:
`make VARIANT=debug`

/home/spoussa/src/iot/zephyr.js/outdir/snapshot/debug/src/zjs_common.c:29:8: error: unknown type name ‘uint8_t’
 static uint8_t init = 0;

poussa, 2017-02-07:
Can we add debug build to travis so we catch these?

grgustaf, 2017-02-08:
I found that commit 10086db introduced the problem. Also you won't see it if you've already built succesfully but can reproduce it again after `make pristine`.

jimmy-huang, 2017-02-08:
submitted patch #697

grgustaf, 2017-02-10:
Just posted PR #707 which has started doing a test build with VARIANT=debug in Travis. This simple sanity check does seem like a good idea; I looked at running it against other modules and I think the surface area is too small to be worth the penalty for slowing down Travis all the time.

I also added a bunch of new tests though, for RAM=, SNAPSHOT=off, sensors, etc. so we have better coverage now.

poussa, 2017-02-13:
Works.

closed 2017-02-13 by poussa