intel/zephyr.js #686: ashell flashing to Arduino 101 fails

poussa, 2017-02-07:
`[spoussa@zephyr.js (master)]$ make DEV=ashell SIZE=256`

completes nicely with

`Creating dfu images... done.`

But flashing fails....

```
[spoussa@zephyr.js (master)]$ make dfu
dfu-util -a x86_app -D outdir/arduino_101/zephyr.bin.dfu
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 8087:0aba
Run-time device DFU version 0011
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0011
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%       147456 bytes
Download done.
state(2) = dfuIDLE, status(0) = No error condition is present
Done!
dfu-util -a sensor_core -D arc/outdir/arduino_101_sss/zephyr.bin.dfu
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 8087:0aba
Run-time device DFU version 0011
Claiming USB DFU Interface...
Setting Alternate Setting #7 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0011
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=======                  ]  29%        40960 bytes failed!
state(10) = dfuERROR, status(8) = Cannot program memory due to received address that is out of range
```

grgustaf, 2017-02-07:
Yeah, as you concluded elsewhere, this is because now everyone should revert to 144KB x86 partition (factory condition). The flashpack to do this is on the [release page](https://github.com/01org/zephyr.js/releases).

Notice that it wrote only ~144KB to the x86 partition, but it's trying to write more than 40K to the arc partition. That's because it expects the original configuration.

Once you downgrade, the only known downside is that the zjs-0.1 and zjs-0.2 branches will still need you to use the old method. I should maybe backport the change to the zjs-0.2 branch.

poussa, 2017-02-08:
Works after reverting back to the 144K default factory configuration.

closed 2017-02-08 by poussa