intel/zephyr.js #687: Buffering of Accelerometer/Gyroscope Data at Higher Sampling Frequencies

Raffprta, 2017-02-07:
Opening up the issue we discussed in IRC, in which an addition to the API should be made to get Accelerometer/Gyroscope data across to application space for use, as best as can be managed within the framework. :-)  

If 1600Hz samples can be buffered and sent, that would be ideal.

grgustaf, 2017-02-28:
@jimmy-huang can correct me if I'm wrong, but I believe he's determined that the A101 can really only report the values at 800Hz. He's merged some changes to do allow you to get all those samples in JS, however if you start doing much processing you will run out of time and start losing data. So I think we still need to offer an API that batches up the samples to reduce context switch overhead and give you more processing time in JS.

If you want to give us any feedback with your real algorithm about what frequency you're able to be successful with, that would be great.

kenchris, 2017-02-28:
We are looking at that for the spec, a kind of BTOB-type (bring your own buffer :)), maybe something along the lines of...

```
let data = Float32Array(...);
onchange = function() { sensor.populate(data); doStuff(); }
```