intel/zephyr.js #693: ashell: LED0 does not work on Arduino 101

poussa, 2017-02-08:
The LED0 (onboard) does not work with the ashell (master, d6f1ce185212e892c73dbacd0b865717600b09c8)

I tried with following samples 

samples/arduino/basics/Blink.js
samples/arduino/digital/Button.js
samples/TrafficLight.js

They do not work with shell but do work without it.

poussa, 2017-02-08:
@brianjjones please take a look asap since we are using the shell during the next event.

kenchris, 2017-02-08:
I had RGB working fine yesterday but after upgrading the GroveLCD example doesn't work for me. (both without ashell)

nagineni, 2017-02-08:
I just tried with samples/ButtonLEDs.js from latest master and it is working fine.

poussa, 2017-02-08:
@kenchris what do you mean by upgrading? From which commit to which?

nagineni, 2017-02-08:
Issue is only with LED0. Onboard LED1 is working fine.

poussa, 2017-02-08:
Yes, as the title says: with **ashell** **LED0** does **not** work. With ashell LED1 works. Without ashell I have not discovered any issues either.

brianjjones, 2017-02-08:
Strange, I saw it working yesterday with ashell.  But I'll take a look this morning and confirm.

poussa, 2017-02-10:
@brianjjones Any news on this one? The issue is still there with the latest master (249f119cf6f88da5a54d1a6669a59631e0092329)

Easy way to test is TrafficLights.js with ashell. The onboard led (green, pin13, LED0) stays off but the other 2 leds blink.

poussa, 2017-02-13:
By the time I connect from the Browser to the device, I see on the console:

```
Hello, ZJS world!

src/zjs_callbacks.c:501 zjs_call_callback():
[ERROR] callback does not exist: 0
```

Don't know if that is relevant or not.

brianjjones, 2017-02-13:
This is my priority today, I should have a fix soon.

grgustaf, 2017-02-13:
I believe the problem here is that our "LED0" is tied to IO13, which in ashell is being used for SPI to talk to the flash filesystem. I'll confirm that by enabling SPI in a regular build and I expect that will break the sample.

If this is right, I don't think there is any way around it and we should probably rearrange our LED names so that LED0 becomes LED2 and has a note of the circumstances limiting its usage.

grgustaf, 2017-02-13:
I did confirm that adding one line to the prj.conf: CONFIG_SPI=y, breaks this in the normal (non-ashell) build. So that is indeed the problem and I don't think there is a solution for ashell since it needs SPI for the filesystem.

poussa, 2017-02-14:
Nothing we can do but document. Once that is done I'll close this bug.

grgustaf, 2017-02-28:
Note to self: document in a101 pins, in ashell, and in future in filesystem/spi docs as well.

grgustaf, 2017-03-29:
Fixed via documentation in #930.

PR #930 (2017-03-30): [docs] Document caveat about conflict between SPI and A101 LED
Also, rename the LEDs so the "bad" LED that breaks when SPI is enabled
is LED2 instead of LED0. Update samples to use the new names or switch
to using LED0 as the default.

Fixes #693.
commit f1275512 (2017-03-30): [docs] Document caveat about conflict between SPI and A101 LED (#930)

Also, rename the LEDs so the "bad" LED that breaks when SPI is enabled
is LED2 instead of LED0. Update samples to use the new names or switch
to using LED0 as the default.

Fixes #693.
closed 2017-03-30 by jimmy-huang