intel/zephyr.js #703: [REG] No output log when running samples and tests on Arduino 101

qiaojingx, 2017-02-09:
When running [samples](https://github.com/01org/zephyr.js/tree/master/samples)(such as [HelloWorld.js](https://github.com/01org/zephyr.js/blob/master/samples/HelloWorld.js), [Console.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/Console.js)) and [tests](https://github.com/01org/zephyr.js/tree/master/tests)(such as [test-a101-pins.js](https://github.com/01org/zephyr.js/blob/master/tests/test-a101-pins.js)) on Arduino 101, no output log when using command `watch screen /dev/ttyUSB0 115200` or `minicom -D /dev/ttyUSB0`.
Once this issue occurred, still no output log after re-flash Arduino 101 board with good build(such as [21915c0](https://github.com/01org/zephyr.js/commit/21915c0c2c96f1ec2b69b90da99545a4a24686a0))

Branch|Build ID|Device|Date|Result
---|---|---|---|---
master|[118679a](https://github.com/01org/zephyr.js/commit/118679ae7d5b451cbeaff04e8e054ef09d1ce4de)|Feb 9, 2017|Arduino101|PASS
master|From [b39016f](https://github.com/01org/zephyr.js/commit/b39016f3885e0c2fbb251df1aa9ecddc29536c29) to [2023adf](https://github.com/01org/zephyr.js/commit/2023adf9405e72fbce13d2900e967f4963fa6b86)|Feb 9, 2017|Arduino101|FAIL
master|From [b39016f](https://github.com/01org/zephyr.js/commit/b39016f3885e0c2fbb251df1aa9ecddc29536c29) to [2023adf](https://github.com/01org/zephyr.js/commit/2023adf9405e72fbce13d2900e967f4963fa6b86)|Feb 9, 2017|FRDM-K64F|PASS

jimmy-huang, 2017-02-09:
Did you guys flash back to the 144K partition on the Arduinos?  If you flash them on the existing 256K partitions, they won't work.

qiaojingx, 2017-02-10:
Thanks a lot, they worked after flashing Arduino 101 back to 144K.
Closed.

closed 2017-02-10 by qiaojingx