intel/zephyr.js #725: [Sensor] Fail to run sensor tests on Arduino

qiaojingx, 2017-02-14:
Fail to run sensor tests([test-accelerometer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-accelerometer.js), [test-ambientlight.js](https://github.com/01org/zephyr.js/blob/master/tests/test-ambientlight.js), [test-gyroscope.js](https://github.com/01org/zephyr.js/blob/master/tests/test-gyroscope.js)) on Arduino 101 with commit 3f8eb85.

Serial logs:
```
Testing gyroscope APIs                                                                          
ipm_console0: 'Sensor core running ZJS ARC support image'                                       
PASS - sensor: be defined                                                                       
FAIL - sensor: state is readonly property                                                       
                                                                                                
src/zjs_sensor.c:134 zjs_sensor_get_state():                                                    
[ERROR] state not set                                                                           
PASS - sensor: callback value for 'onstatechange'                                               
PASS - sensor: 'idle' change to 'activating'                                                    
PASS - sensor: current state as 'activating'                                                    
currentstate: activating                                                                      
                                                                                                
src/zjs_sensor.c:103 zjs_sensor_ipm_send_sync():                                                
[ERROR] FATAL ERROR, ipm timed out                                                              
                                                                                                
src/main.c:226 main():                                                                          
[ERROR] JerryScript: cannot run javascript
```
Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Arduino 101 | 21915c0 | Feb, 2017 | Blocked by #604 
master | Arduino 101 | 3f8eb85 | Feb 13,2017 | FAIL

jimmy-huang, 2017-03-08:
Should be fixed by #791

qiaojingx, 2017-03-13:
Verified with commit eaf24e9 on arduino 101. This issue is not reproducible.
Closed

closed 2017-03-13 by qiaojingx