intel/zephyr.js #727: [ashell] IO10~IO13 are failed as output pins by zephyrJS IDE

cuiyanx, 2017-02-14:
When run test/test-a101-pins.js by zephyr.js IDE, IO10~IO13 are failed as output pins. but they are passed by zephyr JS on a101.

grgustaf, 2017-02-14:
I believe this is inevitable because those pins are being used for SPI to talk to the flash filesystem. However, we do need to document that better. AFAIK in ashell mode those will never be available.

cuiyanx, 2017-02-15:
OK, thanks and closed.

cuiyanx, 2017-06-02:
@grgustaf Please update [/docs/ashell.md](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md) to clarify

brianjjones, 2017-09-25:
This is invalid since ashell requires SPI in order to work, its needed for filesystem.  If you enable SPI, then pins I010 - IO13 become SPI pins, and no longer function as GPIO.  You will see the same thing happen if you were to add require("spi") to you JS and build without ashell.

closed 2017-10-11 by grgustaf

cuiyanx, 2017-10-12:
Fixed and closed

PR #1559 (2017-10-11): [docs] Updating ASHELL docs with note about SPI pins
Fixes #727 - Pins 10-13 can only be used as SPI on ashell
when using the A101.
commit f78e310f (2017-10-11): [docs] Updating ASHELL docs with note about SPI pins (#1559)

Pins 10-13 can only be used as SPI on ashell when using the A101.

Fixes #727