intel/zephyr.js #728: backspace key isn't handled right in ashell

grgustaf, 2017-02-14:
For me, it prints <CTRL> 8 or something and seems to hit Enter, etc.

brianjjones, 2017-02-28:
Works for me, what kind of keyboard?

qiaojingx, 2017-03-01:
This issue can be reproduced when using `minicom -D /dev/ttyACM0` to connect ashell with terminal.
Commit: fc06a11

brianjjones, 2017-03-25:
Can someone take a screenshot of this? I've never seen it.  Thanks.

qiaojingx, 2017-03-28:
It seems only reproduced when using `minicom -D /dev/ttyACM0`. Not reproduced when using `screen /dev/ttyACM0 115200`
[image]

brianjjones, 2017-03-28:
Fix is here https://github.com/01org/zephyr.js/pull/927

qiaojingx, 2017-03-29:
Verified with commit 1c67773 + #927. This issue is not reproducible.
Will close it after patch merged.

closed 2017-04-01 by grgustaf