intel/zephyr.js #735: [Error] Fail to calling require with an invalid module on Linux

qiaojingx, 2017-02-15:
Fail to calling require with an invalid module on Linux when running test-error.js via command `zephyr.js$ ./outdir/linux/release/jslinux tests/test-error.js` with commit 3f8eb85. Not reproduced on Arduino 101 and FRDM-K64F.

Serial log:
```
Testing error types...

PASS - SecurityError works as expected
PASS - NotSupportedError works as expected
PASS - SyntaxError works as expected
PASS - TypeError works as expected
PASS - RangeError works as expected
PASS - TimeoutError works as expected
PASS - NetworkError works as expected
PASS - SystemError works as expected
PASS - calling require with an integer
PASS - calling require with a long name

src/zjs_script.c:17 zjs_read_script():
[ERROR] error opening file

src/zjs_modules.c:154 native_require_handler():
[ERROR] could not read module modules/easterbunny
calling require with an invalid module name: expected error object
FAIL - calling require with an invalid module name
TOTAL: 10 of 11 passed
```

Branch | Target Device | Build ID | Date | Result
---|---|---|---|---
master | Linux | 21915c0 | Feb, 2017 | PASS
master | Linux | 3f8eb85 | Feb 13, 2017 | FAIL

grgustaf, 2017-02-28:
This should be fixed by #774.

qiaojingx, 2017-03-06:
Verified with commit b29dbbc. This issue is not reproducible on Linux. Closed.

closed 2017-03-06 by qiaojingx